VUnit/vunit #357: ModuleNotFoundError: No module named 'colorama'

ikague, 2018-07-23:
i tried to run the example \vhdl\user_guide\ and got the following report :

D:\Playground\vunit-3.7.0\examples\vhdl\user_guide>python run.py
Compiling into vunit_lib: ..\..\..\vunit\vhdl\string_ops\src\string_ops.vhd                       passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\path\src\path.vhd                                   passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\print_pkg.vhd                           passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\codec_builder.vhd                    passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\codec_builder-2008.vhd               passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\codec.vhd                            passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\string_ptr_pkg.vhd                   passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\string_ptr_pkg-body-200x.vhd         passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\ansi_pkg.vhd                            passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\log_levels_pkg.vhd                      passed
COMP96 WARNING COMP96_0119: "The range is 'null range'." "D:\Playground\vunit-3.7.0\vunit\vhdl\logging\src\log_levels_pkg.vhd" 3
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\integer_vector_ptr_pkg.vhd           passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\integer_vector_ptr_pkg-body-200x.vhd passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\file_pkg.vhd                            passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\integer_array_pkg.vhd                passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\integer_array_pkg-body.vhd           passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\queue_pkg.vhd                        passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\queue_pkg-body.vhd                   passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\string_ptr_pool_pkg.vhd              passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\integer_vector_ptr_pool_pkg.vhd      passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\queue_pool_pkg.vhd                   passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\dict_pkg.vhd                         passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\data_types_context.vhd               passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\codec-2008.vhd                       passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\data_types\src\queue_pkg-2008.vhd                   passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\core\src\stop_pkg.vhd                               passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\core\src\stop_body_2008.vhd                         passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\core\src\core_pkg.vhd                               passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\log_levels_pkg-body.vhd                 passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\log_handler_pkg.vhd                     passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\log_handler_pkg-body.vhd                passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\print_pkg-body.vhd                      passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\logger_pkg.vhd                          passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\logger_pkg-body.vhd                     passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\dictionary\src\dictionary.vhd                       passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\run\src\run_types.vhd                               passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\run\src\runner_pkg.vhd                              passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\run\src\run_api.vhd                                 passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\run\src\run.vhd                                     passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\logging\src\log_deprecated_pkg.vhd                  passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\check\src\checker_pkg.vhd                           passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\check\src\checker_pkg-body.vhd                      passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\run\src\run_deprecated_pkg.vhd                      passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\vunit_run_context.vhd                               passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\check\src\check_api.vhd                             passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\check\src\check.vhd                                 passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\check\src\check_deprecated_pkg.vhd                  passed
Compiling into vunit_lib: ..\..\..\vunit\vhdl\vunit_context.vhd                                   passed
Compiling into lib:       tb_example_many.vhd                                                     passed
Compiling into lib:       tb_example.vhd                                                          passed
Compile passed

Traceback (most recent call last):
  File "D:/Playground/vunit-3.7.0\vunit\ui.py", line 761, in main
    all_ok = self._main(post_run)
  File "D:/Playground/vunit-3.7.0\vunit\ui.py", line 802, in _main
    all_ok = self._main_run()
  File "D:/Playground/vunit-3.7.0\vunit\ui.py", line 836, in _main_run
    self._run_test(test_list, report)
  File "D:/Playground/vunit-3.7.0\vunit\ui.py", line 928, in _run_test
    no_color=self._args.no_color)
  File "D:/Playground/vunit-3.7.0\vunit\test_runner.py", line 57, in __init__
    self._stdout_ansi = wrap(self._stdout, use_color=not no_color)
  File "D:/Playground/vunit-3.7.0\vunit\test_runner.py", line 449, in wrap
    from colorama import AnsiToWin32
ModuleNotFoundError: No module named 'colorama'

D:\Playground\vunit-3.7.0\examples\vhdl\user_guide>

Has someone an idea how i can fix it ?

Thank you in advance ;)

Greetings

kraigher, 2018-07-23:
VUnit depends on colorama package. It is installed when installing VUnit with setup.py  or pip install. When running VUnit directly from the repo without installing you need to install colorama manually via pip install.

ikague, 2018-07-23:
@kraigher thank you for the fast response! true, im trying to run it directly from the repo, Are they other dependencies besides the package colorama ?

kraigher, 2018-07-23:
It is the only dependency

closed 2018-07-23 by ikague

1138-4EB, 2018-07-23:
Just for completeness, you can install all the dependencies with [`requirements.txt`](https://github.com/VUnit/vunit/blob/master/requirements.txt):

`pip install -r requirements.txt`

See https://pip.pypa.io/en/stable/user_guide/#id1

Should any additional dependency be added in the future, or should a specific version of a dependency be required, using `requirements.txt` avoids having to track those changes.