VUnit/vunit #358: wishbone master does not follow command order

eschmidscs, 2018-07-23:
Hej

The wishbone master splits all incoming commands into two queues. The process handling the queues then first handles all read commands.
In my current testcase, I write 2048 values in a memory mapped BRAM and then read them back (simple register interace test). This test fails because the read command is executed before the write commands.

IMHO the command execution should happen in the first process, and not be separated with two additional queues.

Additionally, the wishbone master does not know the command wait_for_idle.

eschmidscs, 2018-07-23:
If you agree, I will
- fuse the processes main and request into one
- implement the wait_for_idle by c&p from the axi lite master (does that work the same?)
and create a pull request for that.

slaweksiluk, 2018-07-23:
Hi, thank you for testing Wishobne BFM!
  - WB master has two queues to support requests coalescing for block transfers. I remember I had some problems when I wanted to get it work without them. Of course it would  be better to have it in single process.
  - As a workaround to avoid command reordering please add one clock cycle delay between write_bus() and read_bus() calls. Indeed, this should be fixed (but I think removing requests queues is not essential). Please also share a code example.

PR #372 (2018-08-21): wishbone master does not follow command order
This should fix #358 
I hope I'm not completely failing the guidelines...
closed 2018-08-21 by kraigher