VUnit/vunit #360: bus error response

eschmidscs, 2018-07-23:
God kväll

I have a rather generic point about the bus master implementation.
It is currently not possible to check bus error responses from the tb driver. The AXI4L master simply checks for an ok bresp or throws an error, the wishbone master simply does not support an err signal.

In my world view, a complete test set of a bus slave that might return an error would include a test that actually triggers the error and checks it.
For that, the bus master model must allow to read the bus response somehow.

Is there a specific reason that the bus master package does not allow this?
Or may I make a feature request to support this?

Best regards,
emanuel

kraigher, 2018-07-24:
There is no reason other than no-one has prioritized to implement such support.

kraigher, 2018-07-24:
I can add support for this mechanism unless you want to make a pull request yourself.

eschmidscs, 2018-07-24:
I'd be happy to implement this, but currently time is extremely short.
Maybe in the next two weeks (if my wife allows me to hack something during vacation ;) )

soilheart, 2018-12-03:
Found this issue the other day and realized that I may have partially "fixed" it with #411, which adds support for expecting other statuses than OK in the AXI lite master VC.

eschmidscs, 2018-12-03:
That is indeed what I was missing. At the time of writing more for wishbone than for axi however.
Anyway, it currently doesn't have much attention here ;) (i.e. full project aflame...)

I would even pull the error response into the bus master package, as this is a very common aspect. But maybe there are other plans.

kraigher, 2018-12-03:
I think it is natural for new features like this to start in the bus specific VCI but once we recognize that it is a common feature which can share API it can be moved to the general bus master VCI.