VUnit/vunit #369: functional coverage is not merged in Modelsim coverage merge

krishnan-gopal, 2018-08-08:
While using Modelsim/Questasim, the final merged coverage file is missing the functional coverage despite that each of the individual testcase coverage has it.

I found that this is because of the option "-strip 1"  in vcover merge (in vunit-master/vunit/modelsim_interface.py). Once I removed this option, I can see the functional covergroups in the merged coverage file.

Basically, the "-strip 1" option excludes all the coverage information outside the DUT, i.e the testbench. But, functional coverage is by definition a part of the testbench, unlike code coverage in the DUT.

Therefore, this option needs to be removed or configurable.
If the user wants to suppress coverage information for certain scopes/files, they can do it using disable_coverage instead.

kraigher, 2018-08-13:
I will add a hook to let the user control the merge command.

kraigher, 2018-08-13:
@krishnan-gopal you can check out the `coverage` branch which will be a new interface for coverage to make it more flexible for the user. It is not backwards compatible. Basically the old `--coverage` command line argument is gone and the user has more direct control of the simulator. You should be able to modify the uart example in the following way on this branch to achieve the same effect as the old ``--coverage=bs`` command line argument would have had.

```python
from os.path import join, dirname
from vunit import VUnit

ui = VUnit.from_argv()
ui.add_osvvm()
ui.add_verification_components()

src_path = join(dirname(__file__), "src")

uart_lib = ui.add_library("uart_lib")
uart_lib.add_source_files(join(src_path, "*.vhd"))

tb_uart_lib = ui.add_library("tb_uart_lib")
tb_uart_lib.add_source_files(join(src_path, "test", "*.vhd"))

# The old --coverage=bs flag automatically did this.
ui.set_sim_option("enable_coverage", True)
ui.set_compile_option("modelsim.vcom_flags", ["+cover=bs"])

def post_run(results):
     # The old --coverage=bs also automatically did this
    results.merge_coverage("coverage.ucdb", args=["-strip", "1"])

ui.main(post_run=post_run)
```

It would be nice to get your feedback if this will work for your use case.

krishnan-gopal, 2018-08-14:
@kraigher I will try it out.
Would there still be some way to turn on/off coverage from the Vunit command line ? I think its rather important to have such a switch, since it would result in a severe simulation performance loss to have coverage always enabled.

kraigher, 2018-08-14:
@krishnan-gopal VUnit has always made it easy to add custom command line arguments (https://vunit.github.io/python_interface.html#adding-custom-command-line-arguments). You can add your own `--coverage` flag in just a few lines and set compile options based on that to reflect the different configurations with and without coverage that you want to be able to select via command line.

kraigher, 2018-08-16:
@krishnan-gopal Had any success with this?

krishnan-gopal, 2018-08-17:
I am getting some Python errors, and I need to look into them. I'll get back to you start of next week.

krishnan-gopal, 2018-08-24:
Ok. I have tried this out with my testbenches, and after a few fixes (on account of not being a python expert), it works without any problems.
I like this approach of being able to control the coverage merge more granularly. Thanks!
@kraigher Have you considered doing the coverage merge right after each individual test is finished ? Basically, once the test passes, just merge it right away to the overall coverage database. There are several advantages of doing this.

kraigher, 2018-08-24:
@krishnan-gopal It can be done to parallelize the merge execution time with the test execution time given the machine has free processor cores. It would require a different api since merge cannot be done in a post run callback.

krishnan-gopal, 2018-08-24:
Alright. We can close this issue now.

closed 2018-08-24 by krishnan-gopal

kraigher, 2018-08-24:
@krishnan-gopal So you do not see it as a problem that the individual coverage files are merged post run? You said you saw several advantages.

krishnan-gopal, 2018-09-10:
Sorry for the late reply, I was away on vacation.

The advantages of a post-run merge are:

1. The simulation-generated data for passed testcases can be deleted to save space, instead of accumulating all the data until the very end only for saving coverage data

2. Accumulated coverage progress can be viewed anytime in the middle of the regression (think of regressions that last several days) instead of waiting until the end