VUnit/vunit #373: wishbone master: strobe/stall/ack probabilities

eschmidscs, 2018-08-14:
Hi

While trying to fix the wishbone master command order, I tried to also verify the time required by each command in the testbench.
A proposal for this is here:
https://github.com/eschmidscs/vunit/tree/wbm_time

However, there are two general issues with the current probabilities:
First, they are mathematically unbound. It is possible that the master stalls forever.
Second, the stall should IMHO not be a probability but a queue length. The probability that the queue stalls is given by the ack probability of the slave. If both are probabilities, this becomes extremely hard to verify.

So I propose that the strobe and ack probabilities should rather be the maximum wait cycles for the random generator (i.e. use some rand(0, MAX) to get the number of wait cycles). Further, the stall probability should rather be a queue length.

Currently, only the wishbone and avalon VCs seem to have these mechanisms (thanks, Slawomir). Are there any plans for the other VCs in this direction?

kraigher, 2018-08-14:
@eschmidscs The AXI read/write slaves (https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/axi_slave_pkg.vhd) also support setting a stall probability on each bus channel (AW, AR, R, W, B) as well as a minimum/maximum response latency from AW+W to B as well as AR to R.

eschmidscs, 2018-08-14:
@kraigher Sorry, I managed to look only at the masters... which is the wrong end for the topic.

Without digging too deep: The AXI slaves seem to provide about everything.
What I tried to propose is the min/max latency for the wishbone slave acknowledge. This should not be a simple probability, but a time range (as done for the AXI).

@slaweksiluk The issue with the simple probability is the unbound delay. What's the chance for 100 cycles delay? strobe_prob^100. What's the chance for infinite delay? strobe_prob^infinity - this is not zero.
Written that way it seems a ridiculous concern. However, if I want to verify that the delay is modelled correctly, there is always the chance for an outlier that doesn't match. A probability of 0.5 as in your example is interpreted as every second cycle in average. However, it also allows a case where you have no action for 1000 cycles. For a test case running 64 cycles, this will break the check. If you give a min and max for the latency, there is no such extreme outlier.
Certainly, a check based on the average can still break. However, it is now possible to do hard checks with the min and the max. You cannot do this with a simple probability.

This obviously also applies for the AXI slave probabilities.

Or do I miss something completely here?

kraigher, 2018-08-14:
@eschmidscs The strategy would be to run many times and check the average and see that observed stall probability it is within the theoretical value with some arbitrary selected margin. Since the random seed is always the same the test will not sporadically fail and be robust. 

The verification components typically use OSVVM inside with a fixed random seed. To my knowledge there is no way to create a true random value in VHDL that will differ between runs unless a random seed is injected from outside the simulation via generic. 

Still modifying something about the test sequence might change which value in the random sequence are used and cause the test to fail unintentionally, the likelihood that the average will be outside the margin when running many iterations can still be kept very small.

eschmidscs, 2018-08-15:
@kraigher That is about what the testbench modification proposed above does.

I probably have to rephrase my complaint anyway: If the wishbone_master would implement the same features as the AXI slaves, I would be happy. The AXI slaves have the response latency as min/max parameter and there is a queue depth that causes the stall to activate.
I think these mechanisms are more important that a simple probability. But if the probabilities are there as well, that's fine.

That I keep talking about the verification issue is somewhat picky. 
It was kind of difficult to get this margin right for all test cases and I had to exclude prob_stall because I didn't know how to properly integrate it. So while I agree that you can keep the likelihood of test breaks small, I still find it tedious.

slaweksiluk, 2018-08-16:
What I need i simple real-random test. For this purpose simple ack/strobe/stall probability parameters are enough. I have no problem if you add some parameters to restrict min and max latency or queue depth. However I think it would be nice to keep the option to stimulate ack/strobe/stall as simple uniform probabilities without any bounds. Do you think it is possible?

eschmidscs, 2018-08-17:
I guess that should be possible, it just need a way of disabling these checks. For the minimum it is simple (just set it to zero).
Disabling queue depth and maximum latency might need an enable flag (is that possible in the axi slaves? I'll check...).