VUnit/vunit #377: VCD file output with modelsim.vsim_flags

Zeigren, 2018-08-21:
I've been trying to get ModelSim to output some VCD files but haven't had much luck so far. I'm on Windows 10 using ModelSim 10.5b(the one bundled with Quartus 18.0).

For testing I've been using the verilog uart example, I've simply added this line
```
tb_uart_lib.set_sim_option("modelsim.vsim_flags", ["vcd add *"])
```
to its run.py. Which upon execution will give me an error like this

```
Starting tb_uart_lib.tb_uart_tx.test_send_many_bytes
Output file: vunit_out\test_output\tb_uart_lib.tb_uart_tx.test_send_many_bytes_324f957a68a62f3323566c903f238d45a63b1080\output.txt
# vsim -modelsimini E:/vs/vunit_out/modelsim/modelsim.ini -wlf E:/vs/vunit_out/test_output/tb_uart_lib.tb_uart_tx.test_send_many_bytes_324f957a68a62f3323566c903f238d45a63b1080/modelsim/vsim.wlf -quiet -t ps -onfinish stop tb_uart_lib.tb_uart_tx vcd add * -L vunit_lib -L uart_lib -L tb_uart_lib -g/tb_uart_tx/runner_cfg="active python runner : true,enabled_test_cases : test_send_many_bytes,output path : E::/vs/vunit_out/test_output/tb_uart_lib.tb_uart_tx.test_send_many_bytes_324f957a68a62f3323566c903f238d45a63b1080/,tb path : E::/vs/src/test/,use_color : true"
# Start time: 18:34:14 on Aug 20,2018
# ** Error (suppressible): (vsim-19) Failed to access library 'work' at "work".
# No such file or directory. (errno = ENOENT)
# Error loading design
Error loading design
```
`vcd add *` works when simulating from the ModelSim GUI and will output a dump.vcd file. 
Any thoughts?

kraigher, 2018-08-21:
The `modelsim.vsim_flags` refers to the additional flags you put after the `vsim` command on the same line. Like the `add` is a flag/arguments to the `vcd` command. What you are doing manually in the GUI is not adding a flag to the `vsim` command. The `vsim` command is the one that initialized the simulation and set the top level test bench. 

What you are looking for is likely the `modelsim.init_file.gui` or `modelsim.init_files.after_load` simulation options which you can read about here http://vunit.github.io/python_interface.html?highlight=init_file#simulation-options. They allow you to run any TCL-script you want after the `vsim` command has been executed. FYI the stuff you write in `.do` files and in the modelsim command prompt is just TCL-scripting.

Zeigren, 2018-08-21:
That makes sense, thanks! It works just fine now!

closed 2018-08-21 by Zeigren