VUnit/vunit #380: axis tuser

eschmidscs, 2018-08-22:
Hi

I tried to add TUSER to the axi stream VC. I'm currently stuck with the optional character of the signal.
How would you implement optional parameters in both stream_master_pkg and axi_stream_pkg?

The stream_master_pkg does not know the length of tuser, so having a push_stream function without it fails because it cannot push the correct vector to the message. Or should I modify the protocol if there is no tuser data (i.e. somethimes push 3 values, sometimes 2)?
So I just left stream_master_pkg as is (tuser is axi specific anyway) and made the axi_stream_master drive tuser=zero if the message was not axi specific.

However, on the slave side, this distinction does not work as there is no pop_axi_stream_msg. There is no client function to receive the tuser signal (no pop_axi_stream).
What would be the best way to implement this?

Thanks and best regards,
emanuel

kraigher, 2018-08-23:
You are on the right track. The `tuser` signal does not belong on the generic stream VCI since it is AXI specific. It should be added to AXI stream specific VCI that is implemented by AXI stream master and slave. Since the AXI specific VCI with `tuser` has already been added to the AXI stream master it is only the slave-side remaining.

slaweksiluk, 2018-08-23:
There is some abandoned PR related https://github.com/VUnit/vunit/pull/315 might find it useful

eschmidscs, 2018-08-24:
Thanks for the reasoning and the pointer, I'll try to get that all working.

LarsAsplund, 2018-08-24:
@eschmidscs Remember to also update the [AXI stream protocol checker](https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/axi_stream_protocol_checker.vhd) and its [testbench](https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/test/tb_axi_stream_protocol_checker.vhd) to handle any signal you add. The rules checked are based on this ARM [document](http://infocenter.arm.com/help/topic/com.arm.doc.dui0534b/DUI0534B_amba_4_axi4_protocol_assertions_ug.pdf), section 4.2. Many rules are the same for several signals so hopefully you can just do some copy and paste of the checks already implemented.

eschmidscs, 2018-08-27:
@LarsAsplund Did that. I moved the copy&paste stuff into procedures. I hope that is ok.

PR #385 (2018-08-30): Feature/axis signals
This should close #380. It reuses #315.
Additionally, I added AXI4STREAM_ERRM_TVALID_RESET
closed 2018-08-30 by kraigher