VUnit/vunit #382: Simulating a XIlinx ROM with MIF in Questa/Modelsim

gunnarstjernberg, 2018-08-24:
Hi,

WHen generating a Xilinx ROM IP (Vivado), I can fill the memeory contents with a COE-file. THe resulting simulation outout generates a MIF-file,

In a stand-alone simulation this MIF is automatically read by vsim if placed in the pwd used when running the simulation. However I cannot get taht to work with my VUnit test bench. Is there a method for making sure the VUnit simulation reads the proper MIF and uses it in the simulation?

Best regards,

      Gunnar

kraigher, 2018-08-24:
There was a current solution to this added in #236 which still works. I was not quite happy with the solution and it is **not** documented in the Public API. I probably will add a better solution and document it while keeping the old solution working undocumented for the benefit of those who already use it.

Basically the current solution gives the user access to the path of the simulator working directory in the [pre_config](http://vunit.github.io/python_interface.html?highlight=pre_config#pre-and-post-simulation-hooks) hook under the name `simulator_output_path`. 

With the current solution there is a potential hazard with parallel simulation since all simulator processes shares the same working directory which means if you have two IP:s with identically named .mif files with different contents there can be a race condition when two tests running in parallel could end up writing the same file with different contents. 

My goal is to clean this feature up and add it to the Public API. First I will rename the `pre_config` argument to `simulator_working_directory` to better communicate the meaning. Secondly I will make sure that each parallel simulator process gets its own unique working directory. Using unique working directories a race condition can never happen since conflicting tests either execute sequentially on the same simulator process or in parallel on different simulator processes. In the first case there is exclusivity in time and in the second case there is exclusivity in space.

curiousengineer, 2021-01-03:
Has this issue been resolved as of 2021?

eine, 2021-01-11:
See http://vunit.github.io/py/ui.html#pre-and-post-simulation-hooks.