VUnit/vunit #398: Add possibility to SKIP tests (suggestion)

torerams, 2018-10-01:
I just discovered VUnit and have started using it.
Thanks for a great framework.

One feature that I would find very useful would be the possibility to SKIP tests. I have several test, and some of the tests may take a long time to run. It would be very useful if I could use some condition to SKIP tests. These tests would be marked as "skipped" in the junit/xunit xml file.

Something like this:
```
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_example_many is
  generic (runner_cfg : string);
end entity;

architecture tb of tb_example_many is
begin
  main : process
  begin
    test_runner_setup(runner, runner_cfg);

    while test_suite loop

      if run("test_pass") then
        report "This will pass";

      elsif run("test_possible_skipped") then

        if <some_contidion_to_skip> then
          skip("This test skipped for some reason");
        else
          report "Test run";
        end if;

      end if;
    end loop;

    test_runner_cleanup(runner);
  end process;
end architecture;
```

I know I can select tests to run in the run.py command line, but having the ability to skip tests in the VHDL code fits better in my automated Jenkins testing, and may be useful to other users as well.

torerams, 2018-10-02:
Yes, I want to run fast tests on every commit, and all test (slow test also) every night.

In my example the test is skipped just fine, but the test will be marked as "passed", when it was in fact skipped. Reading up on the VUnit framework it looks like logging, checking etc only has pass/fail result.

I made a not-so-elegant workaround for this that is working for me.

Created a new logging level with a fixed "SKIPPING_TAG" text that I will use later:
```
constant    skip_test                       : log_level_t := new_log_level("SKIPPING_TAG", fg => yellow);

show(display_handler, skip_test);
```
In the testbench I can now "skip" like this:
```
if not(RunSlowTests) then
  log("Skipping Missing DAC test", skip_test);
else
  .....
```

Finally I made a small Python script that is run in the Jenkins job, to do some xunit.xml file post processing. Search for tests with "SKIPPING_TAG" and mark these tests as skipped in the xunit.xml file.
Skipped tests are now showing up as skipped in Jenkins:
[image]

I have a workaround for this, so feel free to close this issue if possibility to skip tests in the VHDL code is not useful for others.

torerams, 2018-10-02:
This solution sounds great. 
The tests disabled with this attribute approach should turn up "N/A" in Jenkins and that is just fine.

I am new to VUnit and still learning the basic stuff, but hope to look into this approach later.
Thanks for the response and help.

joshrsmith, 2018-10-04:
For the record, there was some discussion about skipping tests in #293 

I have run into some very similar difficulties as OP where there are some tests that just take too long to run frequently. Having the ability to find/filter some tests with/without certain attribute/tag so that they can be excluded would be great.

kraigher, 2018-10-04:
Maybe we can add a command line switch for running tests which either have or does not have an attribute set.

# Example
```
# Nigthly tests
> python run.py --with-attribute=.nigthly

# Tests for each commit
> python run.py --without-attribute=.nigthly
```

This would just be another way to filter which test to run and would have the same effect as specifying the names of the tests to run explicitly. Thus tests not run will not end up in the test report.

Attributes are merged to master now and described here: http://vunit.github.io/python_interface.html#id5

kraigher, 2018-10-05:
I pushed a commit now to master that allows test filtering based on existence/absence of attributes in the way I described in my previous comment.

closed 2018-10-05 by kraigher