VUnit/vunit #403: How can i solve "--coverage" problem

sasamy1, 2018-10-22:
I try to generate coverage result. I have tested 2 cases. 
but all of results are wrong.
somebody can answer how to solve this problem?

Case#1  just "--coverage"
Case#2 add option "--coverage s"

my test environment is follow,
- vunit : from master version
- source : vunit-master\examples\vhdl\com
- simulator : RivieraPRO 2016.06
----
Case#1
- command : 'python run.py *publish* --coverage'
- expected result : acdb file is generated
- result : acdb file is not generated
[image]

----
Case#2
- command : 'python run.py *publish* --coverage s'
- expected result : acdb file is generated and test pass
- result : acdb file is generated but test failed
[image]

kraigher, 2018-10-22:
The com example including coverage works with RivieraPro 2017.10 here on Linux. Maybe you have found a RivieraPro bug? Can you try a newer RivieraPro version?

sasamy1, 2018-10-23:
I changed RivieraPro version 2016.06 to 2017.02. and Case#2 is solved. 
Thanks you for answer. 
[image]

but Case#1 still is not solved. is it normal? 
or "--coverage " option need to additional switches like following?
[image]

kraigher, 2018-10-23:
Regarding the `--coverage` flag the arguments are tool specific and for RivieraPRO they are those you mention. For example --coverage=bs would enable branch and statement coverage. Without arguments no coverage was enabled.

Actually as unrelated to this issue we just removed the `--coverage  flag in favor of a more flexible and powerful mechanism where it is up to the user to set exactly the coverage options they want. You can read about it here: http://vunit.github.io/release_notes.html. It is a breaking change since we remove the `--coverage` flag in favor of the new system.

sasamy1, 2018-10-23:
thank you for answer. and i will read release_notes and adopt my project.

closed 2018-10-23 by sasamy1