VUnit/vunit #409: Bug Report: Semicolon in generics string

franz-95, 2018-11-20:
Hey.
So when I have a ';' within a string in the generics, the parser gets confused even though the syntax is right.

kraigher, 2018-11-20:
I will investigate and fix it. In the future VUnit will probably use a real parser from this project instead: https://github.com/kraigher/rust_hdl. It will make these kinds of bugs unlikely. Currently VUnit just uses regex which is not a complete parser and thus has these kinds of problems.

kraigher, 2018-11-20:
I could reproduce the problem and pushed a fix for it. Could you try with the latest master branch and report back if it solved your problem?

kraigher, 2018-11-20:
Will soon part of release 4.0.7.

closed 2018-11-20 by kraigher

franz-95, 2018-11-20:
Thanks for the quick reply and release. I don't have the problem anymore!! :+1: