VUnit/vunit #410: Stack Trace with line numbers

NINI1988, 2018-11-20:
Is it possible to print an error stack trace with line numbers, so that editors like vscode can jump to the specific line?
E.g. `C:/test/tb_can_rx.vhd:236` instead of `C:/test/tb_can_rx.vhd 236`
Notice the colon.

Original stack trace:
```
# Test Run Failed!
#
# Stack trace result from 'tb' command
#  C:/Python37-32/lib/site-packages/vunit/vhdl/core/src/core_pkg.vhd 84 [address ff2a0cfb] Subprogram core_failure
# called from  C:/Python37-32/lib/site-packages/vunit/vhdl/logging/src/logger_pkg-body.vhd 746 [address ff2af123] Subprogram decrease_stop_count
# called from  C:/Python37-32/lib/site-packages/vunit/vhdl/logging/src/logger_pkg-body.vhd 741 [address ff2af1eb] Subprogram decrease_stop_count
# called from  C:/Python37-32/lib/site-packages/vunit/vhdl/logging/src/logger_pkg-body.vhd 918 [address ff2aa6e8] Subprogram log
# called from  C:/Python37-32/lib/site-packages/vunit/vhdl/check/src/checker_pkg-body.vhd 119 [address ff2bbd48] Subprogram failing_check
# called from  C:/Python37-32/lib/site-packages/vunit/vhdl/check/src/check.vhd 3714 [address ff2c9291] Subprogram check_equal
# called from  C:/Python37-32/lib/site-packages/vunit/vhdl/check/src/check.vhd 3665 [address ff2c906e] Subprogram check_equal
# called from  C:/test/tb_can_rx.vhd 236 [address ff43fa64] Process data_check_process
```

kraigher, 2018-11-20:
That output is from the simulator. It is out of control for VUnit.

NINI1988, 2018-11-20:
OK :-(
Thanks for the reply

closed 2018-11-20 by NINI1988