VUnit/vunit #416: How to use vunit_user_init?

jrwagz, 2018-12-06:
I notice in the documentation of the commands available inside vsim (questa/modelsim) a reference to a "user defined init file" however, it's not clear to me how we are supposed to indicate which file this is.  I searched for some documentation on the topic and couldn't find anything.

Can someone clarify this for me? Thanks in advance.

kraigher, 2018-12-06:
The file is set via the `modelsim.init_file.gui` simulation option documented here: http://vunit.github.io/python_interface.html?highlight=init_file.gui

flyinged, 2020-01-27:
> 
> 
> The file is set via the `modelsim.init_file.gui` simulation option documented here: http://vunit.github.io/python_interface.html?highlight=init_file.gui

@kraigher , thanks for the answer, but the link points to a 404. Has the documentation been moved?
Thanks

eine, 2020-01-27:
@flyinged, the section was reorganised. It is now http://vunit.github.io/py/opts.html#simulation-options