VUnit/vunit #417: axi_stream_master: inserting wait_for_time_msg breaks synchronous nature of stream transactions

olafvandenberg, 2018-12-07:
To get 'randomized' time delays between multiple axi streaming master transactions, we insert wait_for_time messages in between axi streaming pushes to the axi stream master verification component.
For our (functional) verification we don't want to model any clock to output delays but want to stay aligned to the edge of the clock.
There is no possibility to do this in the current master.
I've made a workaround by inserting a 'wait until rising_edge(aclk)' after receiving a 'wait_for_time_msg'.
But perhaps it would be nicer to have a wait_for_time variant that aligns to the clock edge in the sync_pkg?

kraigher, 2018-12-07:
I think the the best solution is to make sure the axi_stream_master always align its transactions to the rising_edge of the clock. Thus wait for time can still be fractional clock period but the next transaction will align to the clock. If you are interested in becoming a contributor I could walk you through how to do this with the axi_stream_master. We have already done it with the axi_lite_master.

olafvandenberg, 2018-12-07:
Ok thats sounds good, could you walk me through?

olafvandenberg, 2018-12-07:
found the commit with the changes to the axilite master regarding the clock alignment, I'll have a look at them..

kraigher, 2018-12-07:
Yes you can use that commit as an inspiration. The principle is to run the message passing in one process and the bus signal handling in another process. The message passes process will not be aligned with the clock but the bus signal process will always be aligned with the rising edge of the clock. The message passing process communicates with the bus process via queues which the bus process samples for non-empty condition on every clock edge.

Maybe in the future we can make have some kind of passive message passing support such that the main process can be aligned with the clock as well.

olafvandenberg, 2018-12-07:
I've reworked it in this way, but i have some trouble with running the tests for the vc's.
I get a fail on the axi lite master pkg, i'll have to check out why..
Also I think the way the axi lite master bus process always wait on a rising edge and a none empty queue would prevent the axi streaming master from having a back-to-back transfer, so i think only waiting on an edge after a wait for time message is better.
The approach with the two processes does give the advantage of being able to implement the wait until idle properly, so a queue can be pushed with a lot of transactions at once, and the tb can then wait for them to finish.

olafvandenberg, 2018-12-10:
I would like to share my changes for review, do I need to be added to this repo, or how to go about this?

kraigher, 2018-12-10:
You push it to your fork and then make a pull request to the main repository

kraigher, 2018-12-10:
I guess we can close this now

closed 2018-12-10 by kraigher