VUnit/vunit #421: axi_stream_master: lack of reset

olafvandenberg, 2018-12-11:
The axi stream master does not have a reset, either as an output or an input.
The axi stream protocol checker checks signal states on reset active/non-active, it would be cleaner to add either a reset output to the master or an input, or perhaps let the user choose.
When it is an output it should also be supported by the vci.

kraigher, 2018-12-11:
Yes it is a valid feature. It could have a reset output that is set via command and a reset input that is the default assigned to inactive. The command could override the input reset. Want to make a PR for it?

kraigher, 2018-12-12:
Sounds like both could work. There could also be two signals one in and one out.

olafvandenberg, 2018-12-12:
made a pr, could you check it out?

closed 2018-12-13 by olafvandenberg