VUnit/vunit #423: SYSTEMVERILOG SUPPORT: Warnings while using macros with spaces

krishnan-gopal, 2018-12-12:
SystemVerilog allows the use of spaces to seperate a macro and its arguments, even when there were no spaces in the declaration of the macro. But the Vunit parser doesn't allow this and issues warnings.

In the verilog file below, the macro "MY_MACRO" is defined without spaces before the start of the brackets "(".  In the veriog code where the macro is called, there is a space after the macro and before the start of the brackets. In such a case, a warning is issued by Vunit as shown.

Since the UVM base class library uses this very often, hundreds of such warnings (and similar warnings) are issued during compilation which can be misleading.

> WARNING - Bad `define argument list
>
> at tb_top.sv line 13:
>
>   `MY_MACRO (x1, x2)
>
>   ~~~~~~~~~

// START OF FILE
```
// MACRO DEFINITION - No SPACE BETWEEN `MY_MACRO and (

`define MY_MACRO(ARG1, ARG2) \
  $display(ARG1, " with ", ARG2);

module tb_top();

string x1 = "123";
string x2 = "456";

import vunit_pkg::*;
  `include "vunit_defines.svh"

`TEST_SUITE begin
// WARNING OCCURS DUE TO SPACE BETWEEN `MY_MACRO and (
  `MY_MACRO (x1, x2)
end 

endmodule
```

// END OF FILE

kraigher, 2018-12-13:
Does SystemVerilog allow macros without arguments followed by parenthesis?

```verilog
`define MY_MACRO1 foo
`MY_MACRO1 (x, y) // Expands to foo (x, y)

`define MY_MACRO2(x,y) x+y
`MY_MACRO2 (x, y) // Expands to x+y
```
If so this needs to be distinguished.

krishnan-gopal, 2018-12-13:
Yes, it does. Both the cases in your code above are allowed.

krishnan-gopal, 2018-12-13:
you can use the UVM example testbench (attached below) to test out the parsing. If you run the file vunit_uvm_example/uvm_tb_1/run.py , you will be able to see several Vunit warnings generated from parsing the UVM base classes.
[vunit_uvm_example.zip](https://github.com/VUnit/vunit/files/2675718/vunit_uvm_example.zip)

commit 606fbbf4 (2018-12-14): Fix space before system verilog macro argument. Closes #423
closed 2018-12-14 by kraigher