VUnit/vunit #424: Addition to VC's: SPI master/slave and updated AVMM master/slave

bjacobs1, 2018-12-13:
Y'all,

I've been using vunit for some time now, and I feel like giving back some stuff I build with it to the project. So far, I've build 4 VC's around the SPI and AVMM interfaces. They all support full duplex stream interfaces (e.g. you can call push_stream() and pop_stream() on both the master and the slave), and popping gets synchronized with the availabilitiy of data: received data and requests get queue'd, and if both queue's have data, a reply gets formulated. I've also provided tests.

If you're interested, feel free to see for yourself. If you like what you see, let me know, I'll make a pull request. If you like to discuss some things, or prefer to see stuff changed, let me know as well.

Kind regards,
S. Jacobs

[verification_components.zip](https://github.com/VUnit/vunit/files/2675236/verification_components.zip)

kraigher, 2018-12-13:
Yes I am interested. It would be easiest for me to comment and review if you made a pull request out of it directly.

bjacobs1, 2018-12-20:
I created the pull request last week, and it seemed to pass the tests. Is there anything else that you need for the review?

kraigher, 2018-12-20:
Not really it is just I have not had time to look at it.

LarsAsplund, 2019-02-01:
I'm closing this since we have an open pull request handling the issue

closed 2019-02-01 by LarsAsplund