VUnit/vunit #426: Problem with --without-attributes with more than one attribute

nifx, 2018-12-14:
Hi,
first of all, thank you very much for this tool. I really appreciate your work. It is so cool to use vunit.

I am currently struggling with the command line argument --without-attributes

Here an example:

  p_test: process
  begin
    test_runner_setup(runner, runner_cfg);
    
    while test_suite loop
      if run("test1") then
        -- vunit: .ignore
        wait for 1 ns;
        
      elsif run("test2") then
        -- vunit: .nightly
        wait for 1 ns;
 ...

test1 has the attribute .ignore
test2 has the attribute .nightly

 run.py --without-attributes=.ignore
-> test1 is ignored, test2 is executed -> correct behaviour

 run.py --without-attributes=.nightly
-> test1 is executed, test2  is ignored -> correct behaviour

But now, I want to ignore both tests, so that no test is executed.
I tried several approaches, but without any success:

 run.py --without-attributes=.ignore .nightly
->  test1 is ignored, test2 is executed
-> print(args.without_attributes) -> ['.ignore']

run.py --without-attributes=".ignore .nightly"
-> test1 and test2 executed
-> print(args.without_attributes) -> ['.ignore .nightly']

 run.py --without-attribute=.ignore --without-attribute=.nightly
->  test1 is executed, test2 is ignored

Could you please advice me, what I am doing wrong?
Thanks a lot in advance.

BR Nico

kraigher, 2018-12-14:
There was a problem with the argument parsing not working as intended.

kraigher, 2018-12-14:
You can try the fix on the master branch.

nifx, 2018-12-14:
Ok. Thanks for your quick reply.
I tested it and it works with the call
--without-attributes=.ignore --without-attributes=.nightly

Is this then the intended way to use?
Or is there a way to have only one --with-attributed followed by several attributes?

If not, that's no problem. Just out of interest.

Thanks a lot.

kraigher, 2018-12-14:
Yes that is the intended way which works out of the box with the argparse library. It is for CI scripts and not for interactive use mostly so the extra verbosity doesn't matter so much.

nifx, 2018-12-14:
Yes, that's right.
It's the same on my side, also for CI. So, it doesn't matter.
Thanks again.