VUnit/vunit #427: modelsim gui merge coverage

imd1, 2018-12-14:
After running a test using the Modelsim GUI, the merge coverage fails on line 336 of modelsim_interface.py as "AttributeError: 'NoneType` object has no attribute `teardown`.

It appears that the Modelsim version of merge_coverage() differs from the Rivierapro version (lines 333-335 of riverapro_interface.py) in that it does not have an "if not NONE" protection 

```
        if self._persistent_shell is not None:
            # Teardown to ensure acdb file was written.
            self._persistent_shell.teardown()
```

My question is whether this protection should also be added to the Modelsim version?

kraigher, 2018-12-14:
Yes this was recently fixed in rivierapro and the same fix should be applied in modelsim.py? Could you make a PR?

imd1, 2018-12-14:
Done

closed 2018-12-15 by kraigher