VUnit/vunit #433: Option to not link against a library

eschmidscs, 2018-12-18:
Good evening

Currently, add_external_library always adds its argument to the -L parameter of questa/modelsim. I guess it also always creates a statement in modelsim.ini

I just happend to stumble over this, because I need both unisim and unisims_ver (xilinx), but the first one would actually only be a vhdl library, thus should not be added to -L.
This would not hurt if unisim would not have an "issue" (Xilinx must define if it is a bug) that makes questasim abort when unisim is linked against verilog-based xilinx blocks.

Would it be an option to add a flag to external libraries, if they should be used for linking or not?
Or is this something very questa/modelsim specific?

Best regards,
emanuel

kraigher, 2018-12-19:
I am for adding an option to `add_external_library`. The question is just what terminology to use when exposing the option. Are you able to implement this and make a pull request?

eschmidscs, 2018-12-19:
Not sure if I am able to do this. I’m not very pythonic. If you want me to make an exercise out of it, I can try to do it ;)

What do you mean by terminology? I think I would add that as named parameter only, with a default to do the linking. That would be fully backwards compatible, if I grasp that correctly.
Or is that rather a question about wording?

How does that work for other simulators? Do they have the language-dependent (vhdl through vmap, verilog through -L) mechanism as well?

kraigher, 2018-12-19:
It is matter of wording, what to call the named parameter. RivieraPro had a similar mechanism. Should it be called "link" or "link_verilog". If it is just verilog that is affected then maybe link verilog is a better name.

eschmidscs, 2018-12-19:
Checking the reference manual for questasim:
- vcom does not have -L
- vlog references the chapter "Verilog Resource Libraries" in the users guide.
I do believe that it is just a verilog thing.

The documents call that search path. There is actually a modelsim.ini variable LibrarySearchPath that applies to vlog and vsim. It is identical to -L.

So maybe "add_verilog_search_path"? Too long?

kraigher, 2018-12-19:
Both the vlog and the vsim commands have this flag. What does the vsim command say about the flag?
Strange that they call it search path when the argument is a logical library name like "unisim" and not a file system path.

Maybe `add_to_verilog_search_path` is a good name. It does not matter that it is long as this is not used in many places and it is better that it is self-documenting.

Code:
https://github.com/VUnit/vunit/blob/master/vunit/modelsim_interface.py#L163-L164
https://github.com/VUnit/vunit/blob/master/vunit/modelsim_interface.py#L245-L246
https://github.com/VUnit/vunit/blob/master/vunit/rivierapro_interface.py#L155-L156
https://github.com/VUnit/vunit/blob/master/vunit/rivierapro_interface.py#L155-L156

eschmidscs, 2018-12-19:
Copy from questa reference manual, vsim command:
-L <library_name> …
(optional) Specifies the library to search for top level design units instantiated from Verilog,
and for VHDL default component binding. Prints a list of all visible top level libraries, if a
top level design unit cannot be found. Refer to “Library Usage” in the User’s Manual for
more information. If you specify multiple libraries, each must be preceded by the -L
argument. Libraries are searched in the order in which they appear on the command line.

What means that it actually has a meaning for VHDL when it comes to the top level entity.

vopt is a lot shorter:
• -L <libname>
(optional) Searches the specified resource library for precompiled modules. The library
search you specify here must also be specified when you run the vsim command.

I guess vsim itself only uses -L for the top level. However, it will pass the options to vopt if that is not called by itself, thus both descriptions apply.