VUnit/vunit #436: Code Coverage Automated Generation

dsp20, 2019-01-16:
I was trying to generate coverage report with Vunit, Python, Questasim. With the older release of Vunit I was able to generate Coverage report for module under test (and not the test bench coverage report) with the command,
 
Python run.py –coverage bcesf
And then executing command,   
vsim -c -do "coverage save -onexit -assert -directive -cvg -codeAll -instance sim:/cam_mux_tb/UUT vunit_out/modelsim/merged_coverage.ucdb
Above command gives me coverage database for specific module. Highlighted part is where I was selecting the module for which I want to generate coverage report. 
 
But with the recent release I have added options in python script as shown in your coverage example code. And then I am using above vsim command to generate specific module coverage report. But that is not working for it. I am getting coverage report only for test bench module and not for specific module. 
 
Please help me how can I generate coverage report for specific module with recent Vunit release.

dsp20, 2019-01-16:
In a recent release, there is no direct coverage option which was previously there as                              
 python run.py --coverage bcesf
Now I need to cover this option in a python script. But I think it is giving coverage only for testbench and not for the unit under test.

kraigher, 2019-01-16:
Might be related to the removal of built in "strip 1" from the merge command https://github.com/VUnit/vunit/issues/369. The user have to add it themselves now.

dsp20, 2019-01-16:
Where can I find details about "strip 1"

kraigher, 2019-01-16:
The questa documentation for the coverage merge command should give you more information. You can add custom flags to this questa command via argument to the Python API merge function.

dsp20, 2019-01-16:
Adding that not working though. 
"coverage save -onexit -assert -directive -cvg -codeAll -instance sim:/cam_mux_tb/UUT vunit_out/modelsim/merged_coverage.ucdb"
How Can I set options like above (coverage save) from vunit

dsp20, 2019-01-16:
Working with "strip" option.
Closing the issue

closed 2019-01-16 by dsp20