VUnit/vunit #437: ghdl code coverage in multi tests

qarlosalberto, 2019-01-17:
Hello,

I am trying to do code coverage with multiple generated tests with ghdl (gcc backend). I have a problem with gcov outputs. One test overwrites the gcov outputs of another.

I have found a solution, change the environment variable for each test. It is possible?

https://gcc.gnu.org/onlinedocs/gcc/Cross-profiling.html

kraigher, 2019-01-21:
Each test run is assigned its own output path even if it is a generated test. In the cases you mention GHDL might need an extra flag to make the coverage and psl artifacts end up in the output path. Alternatively running the GHDL command with the output path as the working directory. The relevant file is ghdl_interface. py if you want to make a change.

qarlosalberto, 2019-01-21:
Thanks for answering.

For psl report I have ```ui.set_sim_option("ghdl.sim_flags"      , ["--psl-report=psl_coverage.json"])```
How can I pass the output path?

kraigher, 2019-01-21:
You cannot access the output path and use it in a sim option currently.