VUnit/vunit #438: Test case with certain special characters in test name fail

CherryMan, 2019-01-20:
vunit: 0.1.3
python: 3.6.8

The tests seem to be failing in one of two different ways. To see that, compare the error output of
the test for `~` vs `,`. Some characters work when there are other characters around it (like `~`), while others fail in every case (like `,`).

Following are the test summary and the testbench.
```
==== Summary ==========================
pass lib.tb_test.   (0.6 seconds)
pass lib.tb_test.!  (0.1 seconds)
pass lib.tb_test.#  (0.1 seconds)
pass lib.tb_test.%  (0.1 seconds)
pass lib.tb_test.&  (0.1 seconds)
pass lib.tb_test.'  (0.1 seconds)
pass lib.tb_test.(  (0.1 seconds)
pass lib.tb_test.)  (0.1 seconds)
pass lib.tb_test.*  (0.1 seconds)
pass lib.tb_test.+  (0.1 seconds)
pass lib.tb_test.-  (0.1 seconds)
pass lib.tb_test..  (0.1 seconds)
pass lib.tb_test./  (0.1 seconds)
pass lib.tb_test.<  (0.5 seconds)
pass lib.tb_test.=  (0.1 seconds)
pass lib.tb_test.>  (0.1 seconds)
pass lib.tb_test.?  (0.1 seconds)
pass lib.tb_test.@  (0.1 seconds)
pass lib.tb_test.]  (0.1 seconds)
pass lib.tb_test.^  (0.1 seconds)
pass lib.tb_test._  (0.1 seconds)
pass lib.tb_test.`  (0.1 seconds)
pass lib.tb_test.|  (0.1 seconds)
pass lib.tb_test.~  (0.1 seconds)
fail lib.tb_test.$  (0.1 seconds)
fail lib.tb_test.,  (0.1 seconds)
fail lib.tb_test.:  (0.1 seconds)
fail lib.tb_test.;  (0.3 seconds)
fail lib.tb_test.[  (0.1 seconds)
fail lib.tb_test.\\ (0.1 seconds)
fail lib.tb_test.{  (0.1 seconds)
fail lib.tb_test.}  (0.1 seconds)
=======================================
pass 24 of 32
fail 8 of 32
=======================================
```

```verilog
`include "vunit_defines.svh"

module tb_test;
    `TEST_SUITE begin
        `TEST_CASE("");
        `TEST_CASE("!");
        `TEST_CASE("#");
        `TEST_CASE("$");
        `TEST_CASE("%");
        `TEST_CASE("&");
        `TEST_CASE("'");
        `TEST_CASE("(");
        `TEST_CASE(")");
        `TEST_CASE("*");
        `TEST_CASE("+");
        `TEST_CASE(",");
        `TEST_CASE("-");
        `TEST_CASE(".");
        `TEST_CASE("/");
        `TEST_CASE(":");
        `TEST_CASE(";");
        `TEST_CASE("<");
        `TEST_CASE("=");
        `TEST_CASE(">");
        `TEST_CASE("?");
        `TEST_CASE("@");
        `TEST_CASE("[");
        `TEST_CASE("\\");
        `TEST_CASE("]");
        `TEST_CASE("^");
        `TEST_CASE("_");
        `TEST_CASE("`");
        `TEST_CASE("{");
        `TEST_CASE("|");
        `TEST_CASE("}");
        `TEST_CASE("~");
    end
endmodule
```

LarsAsplund, 2019-01-31:
@CherryMan Is your main concern that some characters fail or that you don't have an error message saying that you're using an invalid name?

CherryMan, 2019-01-31:
There's no documentation (that I found) or error message suggesting that some characters may be illegal, so I assumed that this was a bug.