VUnit/vunit #443: operator "&" is overloaded error in most recent version

GlenNicholls, 2019-02-05:
I have just installed VUnit using `python setup.py install` after running `git clone --recursive https://github.com/VUnit/vunit.git` and am failing to run all example test runs that include OSVVM. After reviewing the error, I see this [GHDL issue for OSVVM](https://github.com/OSVVM/OSVVM/issues/22), however the VUnit repository appears to have the correct version linked as a submodule in the repository. Is anyone able to replicate this and know how to fix this error? I am seeing this output:

```
$ python run.py
Compiling into lib:       src\sobel_x.vhd                                                                                                passed
Compiling into vunit_lib: ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\string_ops\src\string_ops.vhd                       passed
Compiling into vunit_lib: ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\path\src\path.vhd                                   passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\VendorCovApiPkg.vhd                           passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\TranscriptPkg.vhd                             passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\TextUtilPkg.vhd                               passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\NamePkg.vhd                                   passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\OsvvmGlobalPkg.vhd                            passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\AlertLogPkg.vhd                               passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\TbUtilPkg.vhd                                 passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\SortListPkg_int.vhd                           passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\ResolutionPkg.vhd                             passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\RandomBasePkg.vhd                             passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\RandomPkg.vhd                                 passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\MessagePkg.vhd                                passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\MemoryPkg.vhd                                 passed
Compiling into osvvm:     ..\..\..\..\..\..\..\Python27\lib\site-packages\vunit\vhdl\osvvm\CoveragePkg.vhd                               failed
=== Command used: ===
"C:\Program Files\ghdl-0.33\bin\ghdl" -a "--workdir=C:\Users\<user>\Desktop\vunit\examples\vhdl\array\vunit_out\ghdl\libraries\osvvm" --work=osvvm --std=08 "-PC:\Users\<user>\Desktop\vunit\examples\vhdl\array\vunit_out\ghdl\libraries\vunit_lib" "-PC:\Users\<user>\Desktop\vunit\examples\vhdl\array\vunit_out\ghdl\libraries\osvvm" "-PC:\Users\<user>\Desktop\vunit\examples\vhdl\array\vunit_out\ghdl\libraries\lib" C:\Python27\lib\site-packages\vunit\vhdl\osvvm\CoveragePkg.vhd

=== Command output: ===
C:\Python27\lib\site-packages\vunit\vhdl\osvvm\CoveragePkg.vhd:1477:33: operator "&" is overloaded
C:\Python27\lib\site-packages\vunit\vhdl\osvvm\CoveragePkg.vhd:1477:33: possible interpretations are:
*std_standard*:1:1: function "&" [string, character return string]
*std_standard*:1:1: function "&" [string, string return string]

Compile failed
```

For reference, this is also happening when I try to install using `pip install vunit_hdl` instead.

LarsAsplund, 2019-02-05:
What do you get if you do `python run.py --version`? What is your GHDL version?

GlenNicholls, 2019-02-06:
Okay @LarsAsplund, I updated to 0.35 and am able to successfully simulate the examples. It appears that it was the lack of 2008 support in previous versions that are now fixed.

LarsAsplund, 2019-02-06:
Great!

closed 2019-02-06 by LarsAsplund