VUnit/vunit #447: check_relation() with enable_check_preprocessing() does not compile with std_ulogic

el-coderon, 2019-02-21:
Hi, 

I´d like to use the check_relation() with the prj.enable_check_preprocessing() config to get "got-expected"-messages on failure (as mentioned at [https://vunit.github.io/check/user_guide.html#](https://vunit.github.io/check/user_guide.html#) )

Therefore my files have the following lines:

run.py:
```
prj = VUnit.from_argv()
prj.enable_check_preprocessing()
```

tb_mytb.vhd
```

signal nled_a_green_sl_o_s           	: std_ulogic := '1';
signal nled_a_red_sl_o_s             	: std_ulogic := '1';
signal led_a_sim_uslv_s                	: std_ulogic_vector(1 downto 0);    -- for checker only
...
led_a_sim_uslv_s <= nled_a_green_sl_o_s & nled_a_red_sl_o_s;
...
check_relation(led_a_sim_uslv_s = "10");
```

But I do get the error at compilation time:

> Subprogram "to_string" is ambiguous.

But it also tells me:

> Feasible subprograms are
> ...
> (implicit) std_logic_1164.to_string[STD_ULOGIC_VECTOR return std.STANDARD.STRING] 

I think that is quite what I have. What do I wrong?

bradleyharden, 2019-02-22:
Change `"10"` to a qualified expression, i.e. `std_logic_vector'("10")`. The preprocessor-generated code is ambiguous, because `"10"` could be many different types. There might be a way to improve the preprocessor, but this is a simple solution.

el-coderon, 2019-02-22:
Thanks a lot. That works fine!

el-coderon, 2019-02-22:
Oh, I was too fast. First step works fine: It compiles without errors :-)

But there is still no "got <> expected <>" message on check error... Is there any workaround to get this? (as mentioned above: I´ve set the prj.enable_check_preprocessing() )

bradleyharden, 2019-02-23:
Got vs expected usually comes from check_equal. check_relation uses left vs right.

Is it giving you an error message? If so, what is it? You should post more context for your problem. What you've provided isn't really enough to figure out what's going on.

Also, try to reduce the problem to a minimal possible example, and then post that example in its entirety.

bradleyharden, 2019-02-24:
Also, why are you using `check_relation` here? It seems like `check_equal` would be more appropriate.

el-coderon, 2019-02-25:
Hi, thanks for that advice!
The `check_equal` was my first approach. But it didn´t work well. That´s why I tried it with `check_relation`.  Now I tried it again. And it runs fine. (Don´t know what I changed meanwhile).

At the moment I´m fine with this way to do so. However, one of the maintainer should decide if this can be called 'solved' or if `check_relation` should work fine (wiht got<> expected <>) for this, as well .

LarsAsplund, 2019-02-26:
This preprocessor is quite simple and improving it to solve this would require that we figure out the type of the signal which is a much more complex task. For that reason I will close this issue.

closed 2019-02-26 by LarsAsplund