VUnit/vunit #452: Unrecognized extension

GlenNicholls, 2019-03-05:
I am migrating some work code to VUnit and am running into this issue:

    $ python run.py
    Traceback (most recent call last):
      File "run.py", line 70, in <module>
        lib[library].add_source_files(file.replace('\\','/'))
      File "C:\Python27\lib\site-packages\vunit\ui.py", line 1196, in add_source_files
        for file_name in file_names])
      File "C:\Python27\lib\site-packages\vunit\ui.py", line 1222, in add_source_file
        file_type = file_type_of(file_name)
      File "C:\Python27\lib\site-packages\vunit\project.py", line 966, in file_type_of
        raise RuntimeError("Unknown file ending '%s' of %s" % (ext, file_name))
    RuntimeError: Unknown file ending '.ahd' of U:/fw_dev/dsss/support/common/firmware_common-trunk/firmware/src/DualPort_MW.ahd

Unfortunately, we target both Altera and Xilinx, so we deliminate certain IP using .ahd for altera VHDL source and .xhd for Xilinx VHDL source and our build scripts grab the correct IP when we kick off builds. How do I work around this error so I don't have to re-architect our build scripts and file naming convention? Removing these files from simulation source lists is also not practical since I need to expand those simulations to VUnit at some point in the near future.

kraigher, 2019-03-05:
Use file_type="vhdl" in the add_source_file call when auto detection based on file ending is not possible.

GlenNicholls, 2019-03-08:
Perfect, thank you for this. I have used my file tree parser to catch these build specific extensions to instantiate the correct `file_type` for VUnit.

closed 2019-03-09 by LarsAsplund