VUnit/vunit #457: Problem with vunit_restart when using OSVVM and Modelsim 10.6a

aodling, 2019-03-13:
Hi!

I've been using Vunit for a while and just started adding OSVVM features into it. I'm running into some troubles when doing vunit_restart when i have OSVVM objects instansiated. This is when debugging, using the GUI open. 
For example, when having random-generators and scoreboards instansiated and doing vunit_restart, it SOMETIMES (not always) generates a signo 11.
This never happens the first time I start the simulation, just on restart. 
I'm just wondering if anyone else has seen anything like this?

Regards,
Andreas

kraigher, 2019-03-13:
Sounds like a bug in modelsim that should be reported to mentor graphics. Probably easier for them if you can give them a mimimal reproducer.

LarsAsplund, 2019-05-31:
@aodling Any updates on this? It's hard for us to proceed without a minimal reproducible example.

LarsAsplund, 2019-12-23:
I'll close this now. We can't reproduce it and it sounds like a simulator problem rather than a VUnit problem.

closed 2019-12-23 by LarsAsplund