VUnit/vunit #460: Add code coverage support ActiveHDL

vmfox, 2019-03-21:
Following my recent chat on Gitter regarding cove coverage support for ActiveHDL I would like to contribute the changes that would allow this to happen. I have a working version of the relevant .py file which was tested on a real project, and I can prove that the modification generate coverage files for each unit and final merged coverage file. The run.py interface is identical to the already existing one and hence would not require any changes.

I am not able to branch off to add this feature so it would be great if someone could assist me with it.

kraigher, 2019-03-21:
The workflow of creating PRs is explained by github here: https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

vmfox, 2019-03-21:
Tack så mycket.

kraigher, 2019-03-30:
I guess we can close this issue as the PR is merged.

closed 2019-03-30 by kraigher