VUnit/vunit #463: modelsim.init_files.after_load not working as expected

jrwagz, 2019-04-02:
I am attempting to set some simulator variables via the modelsim.init_files.after_load script option, and the variables that I set are not remaining set afterwards.  Here are the specifics:

In my run.py file I add the following line (before ui.main() is called)
```ui.set_sim_option('modelsim.init_files.after_load',['sim_init.tcl'])```

The contents of sim_init.tcl are simple:

```echo "DEBUG - RUNNING SIM_INIT.TCL"

set StdArithNoWarnings 1
set NumericStdNoWarnings 1

echo "StdArithNoWarnings = $StdArithNoWarnings"
echo "NumericStdNoWarnings = $NumericStdNoWarnings"```

When the simulator launches the GUI, or if I run in non-GUI mode, I can see that the script is being called, and that the variables are getting set in the script.

```# DEBUG - RUNNING SIM_UTILS.TCL
# StdArithNoWarnings = 1
# NumericStdNoWarnings = 1```

However, when the simulation runs, these variables are set back to 0.  And even if I manually run the _vunit_source_init_files_after_load command from the GUI console, I can see that the variables aren't set immediately after calling it:

```
echo $StdArithNoWarnings
# 0
echo $NumericStdNoWarnings
# 0
_vunit_source_init_files_after_load
# Sourcing file /data/engtmp/jrwagner/cnes_fpga/git_cores/cswip_cnes_cores/sim_vunit/sim_init.tcl from modelsim.init_files.after_load
# DEBUG - RUNNING SIM_INIT.TCL
# StdArithNoWarnings = 1
# NumericStdNoWarnings = 1
# 0
echo $StdArithNoWarnings
# 0
echo $NumericStdNoWarnings
# 0
```

Any idea what is going on here?

jrwagz, 2019-04-02:
Questa/Modelsim version:

```
# //  Questa Sim-64
# //  Version 10.5c linux_x86_64 Jul 20 2016
# //
# //  Copyright 1991-2016 Mentor Graphics Corporation
# //  All Rights Reserved.
```

jrwagz, 2019-04-02:
Furthermore, if I directly source the sim_init.tcl file in the GUI console, I get the expected behavior:

```
echo $StdArithNoWarnings
# 0
_vunit_source_init_files_after_load
# Sourcing file /data/engtmp/jrwagner/cnes_fpga/git_cores/cswip_cnes_cores/sim_vunit/sim_init.tcl from modelsim.init_files.after_load
# DEBUG - RUNNING SIM_INIT.TCL
# StdArithNoWarnings = 1
# NumericStdNoWarnings = 1
# 0
echo $StdArithNoWarnings
# 0
source ../../sim_init.tcl
# DEBUG - RUNNING SIM_INIT.TCL
# StdArithNoWarnings = 1
# NumericStdNoWarnings = 1
echo $StdArithNoWarnings
# 1
```

jrwagz, 2019-04-02:
Wow, looks like I was trying to re-implement a feature that was already there!  Once I added this to my run.py I get the behavior I want:

```
sim.set_sim_option(‘disable_ieee_warnings’,1)
```

closed 2019-04-02 by jrwagz

kraigher, 2019-04-03:
You found the feature already. Your original solution would have worked if you added a global declaration for the variables.

```tcl
global foo
set foo 1
```