VUnit/vunit #464: Invalid comile order

slaweksiluk, 2019-04-03:
I'm experimenting with generic `memory_pkg. vhd` and it seems compile order is not resolved properly:
```
Compiling into vunit_lib: ../logging/src/logger_pkg.vhd                          passed
Compiling into vunit_lib: src/memory_types.vhd                                   passed
Compiling into vunit_lib: src/memory_pkg.vhd                                     passed
/home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg.vhd:98:40:warning: declaration of "num_bytes" hides function "num_bytes" [-Whide]
/home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg.vhd:104:40:warning: declaration of "num_bytes" hides function "num_bytes" [-Whide]
/home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg.vhd:133:28:warning: declaration of "num_bytes" hides function "num_bytes" [-Whide]
Compiling into vunit_lib: src/memory_pkg_default.vhd                             failed
=== Command used: ===
/home/slawek/install_prefix/bin/ghdl -a --workdir=/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/vunit_lib -P/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/osvvm /home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg_default.vhd

=== Command output: ===
/home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg_default.vhd:24:11: cannot find package body of package "memory_pkg"
/home/slawek/git/github/vunit/vunit/vhdl/verification_components/src/memory_pkg_default.vhd:31:14: package "memory_pkg_default" was not analysed
```
Shouldn't `memory_pkg-body.vhd` be analysed before `memory_pkg_default.vhd `?
Reproducer https://github.com/slaweksiluk/vunit/tree/dep-bug 
Tested on ghdl

umarcor, 2019-04-03:
I found VUnit to have problems resolving the compilation order when genric packages are used; specially when they are defined in some lib, instantiated in a different lib and used in a component compiled to a third lib. The solution I found was to call `add_source_files` once for each file in a correct order. It is just a workaround, but at least it allows to use other VUnit features.

kraigher, 2019-04-03:
The standard is not clear on if a user of package X depends on the body of package X. Normally there is no such dependency. For Riviera-PRO this dependency exists and thus VUnit dependency scanner supports it. For other simulator we do not enable this dependency. The dependency can be enabled by adding a class variable to the corresponding simulator interface. For example in rivierapro_interface.py it is enabled here: https://github.com/VUnit/vunit/blob/master/vunit/rivierapro_interface.py#L36. If the same line was added to ghdl_interface.py it could resolve this problem.

umarcor, 2019-04-04:
@kraigher, thanks for the clarification. Would it be harmful in any context to enable it by default?

kraigher, 2019-04-04:
Probably should be default given this issue. It has previously not been known to be required. Recompilation is avoided when not enabled so that is why it was only enabled for riviera today.

slaweksiluk, 2019-04-05:
It must be related to generic package - Ghdl wouldn't complain about missing body for memory_pkg if it wasn't generic. I will create issue at ghdl repo about that.

umarcor, 2019-04-05:
@slaweksiluk, could you please leave this issue open until the new issue is opened at ghdl/ghdl, or until it is fixed in VUnit directly?

kraigher, 2019-04-05:
Based on the answer from GHDL maintainer we either wait for their fix or add the extra dependency also for GHDL and not only riviera.

slaweksiluk, 2019-04-05:
I thought Tristan does not yet know about this issue. So it is up to you if fix it at Vunit python scripts. Anyway, it is not urgent, as there are 2 or more issues in generic packages support to fix in ghdl before I could use it as backed for memory_pkg.vhd. I can publish my results as PR if you wish.

umarcor, 2019-04-05:
@slaweksiluk, I think that @kraigher wanted *based on the answer from GHDL maintainer* to sound as a future event, not something that has already happened. I comment it, because at first I thought that Tristan had replied already.

kraigher, 2019-04-05:
Yes I was referring to a future event. If GHDL maintainer says this is a bug we do not need to enable package body dependencies for GHDL and we can save recompilation. If it is not a bug then we should enable this extra dependency just like we do with RiveraPRO.