VUnit/vunit #474: Unable to compile Reed Solomon Decover v9 *_sim_netlist.vhdl file from Vivado

jrwagz, 2019-04-12:
When creating IP for a design in Vivado, you have the option to create a file caled <ip_name>_sim_netlist.vhdl that is normally fed directly into your simulator to provide a simulation model of the IPs behavior.

I am currently unable to figure out how to get a specific *_sim_netlist.vhdl file to properly compile and simulate under VUnit.  The name of the IP that I'm trying to simulate is called the "Reed-Solomon Decoder v9" https://www.xilinx.com/support/documentation/ip_documentation/rs_decoder/v9_0/pg107-rs-decoder.pdf

I'm getting a warning in compilation, that I believe shows the crux of the problem, it shows:
```
python3 ./run.py lib.top_tb.check_outputs
WARNING - ..../rs_decoder_7_3_v9_sim_netlist.vhdl: failed to find a primary design unit 'rs_decoder_7_3_v9_rs_decoder_v9_0_12_viv' in library 'lib'
Compiling into lib:       rs_decoder_7_3_v9_sim_netlist.vhdl                 passed
Compiling into lib:       rs_decoder.vhd                                     passed
Compiling into vunit_lib: ../python_scripts/vunit/vunit/verilog/vunit_pkg.sv passed
Compiling into lib:       top_tb.sv                                          passed
Compile passed

Starting lib.top_tb.check_outputs
```

In some simulations, the erroneous behavior is that the module simply always has undefined outputs, but in other designs I'm not able to issue a vunit_load and it complains about the modules who are instancing the IP that is missing an underlying module.

Any ideas on this one?

I've seen the example at https://github.com/VUnit/vunit/tree/master/examples/vhdl/vivado/ which shows how to launch vivado and then generate the products that VUnit needs, however I would really like to avoid launching vivado in my simulation environment, since we normally submit these *_sim_netlist.vhdl files to version control when the IPs are changed, and then our old simulation setup (based on modelsim) was able to simulate them just fine, but I can't get this one to work in VUnit, I assume there may be something unique about it, and that it's not a general problem with all *_sim_netlist.vhdl files, however I don't know that for sure.

My run.py file looks like this:

```
from os.path import join, dirname
from vunit.verilog import VUnit

root = dirname(__file__)

ui = VUnit.from_argv()
lib = ui.add_library("lib")

lib.add_source_files("top_tb.sv")
lib.add_source_files("rs_decoder.vhd")
lib.add_source_files("rs_decoder_7_3_v9_sim_netlist.vhdl")

ui.main()
```

The top_tb.sv file looks like this:
```
library work::*;

`timescale 1ns/1ps

`include "vunit_defines.svh"

module top_tb;
  

  bit clk;
  wire out_dv;

  assign #5ns clk = !clk;

  /****************************************************************************
   * DUT
   ***************************************************************************/
  rs_decoder dut (
    i_clk                   (clk),
    o_rd_en                 (),
    i_dv                    ('b1),
    i_eof                   ('b1),
    i_data                  ('hFF),
    i_rdy                   ('b1),
    o_dv                    (out_dv),
    o_eof                   (),
    o_data                  (),
[... 115 lines cut ...]
# ** Error: (vopt-7052) ..../top_tb.sv(23): Failed to find 'i_data' in hierarchical name '/i_data'.
# ** Error: (vopt-7052) ..../top_tb.sv(23): Failed to find 'i_data' in hierarchical name '/i_data'.
# ** Error: (vopt-7052) ..../top_tb.sv(24): Failed to find 'i_rdy' in hierarchical name '/i_rdy'.
# ** Error: (vopt-7052) ..../top_tb.sv(24): Failed to find 'i_rdy' in hierarchical name '/i_rdy'.
# ** Error: (vopt-7052) ..../top_tb.sv(25): Failed to find 'o_dv' in hierarchical name '/o_dv'.
# ** Error: (vopt-7052) ..../top_tb.sv(25): Failed to find 'o_dv' in hierarchical name '/o_dv'.
# ** Error: (vopt-7052) ..../top_tb.sv(26): Failed to find 'o_eof' in hierarchical name '/o_eof'.
# ** Error: (vopt-7052) ..../top_tb.sv(26): Failed to find 'o_eof' in hierarchical name '/o_eof'.
# ** Error: (vopt-7052) ..../top_tb.sv(27): Failed to find 'o_data' in hierarchical name '/o_data'.
# ** Error: (vopt-7052) ..../top_tb.sv(27): Failed to find 'o_data' in hierarchical name '/o_data'.
# ** Error: (vopt-7052) ..../top_tb.sv(28): Failed to find 'i_stat_ready' in hierarchical name '/i_stat_ready'.
# ** Error: (vopt-7052) ..../top_tb.sv(28): Failed to find 'i_stat_ready' in hierarchical name '/i_stat_ready'.
# ** Error: (vopt-7052) ..../top_tb.sv(29): Failed to find 'o_stat_data' in hierarchical name '/o_stat_data'.
# ** Error: (vopt-7052) ..../top_tb.sv(29): Failed to find 'o_stat_data' in hierarchical name '/o_stat_data'.
# ** Error: (vopt-7052) ..../top_tb.sv(30): Failed to find 'o_stat_valid' in hierarchical name '/o_stat_valid'.
# ** Error: (vopt-7052) ..../top_tb.sv(30): Failed to find 'o_stat_valid' in hierarchical name '/o_stat_valid'.
# ** Error: (vopt-7052) ..../top_tb.sv(31): Failed to find 'o_input_last_missing' in hierarchical name '/o_input_last_missing'.
# ** Error: (vopt-7052) ..../top_tb.sv(31): Failed to find 'o_input_last_missing' in hierarchical name '/o_input_last_missing'.
# ** Error: (vopt-7052) ..../top_tb.sv(32): Failed to find 'o_input_last_unexpected' in hierarchical name '/o_input_last_unexpected'.
# ** Error: (vopt-7052) ..../top_tb.sv(32): Failed to find 'o_input_last_unexpected' in hierarchical name '/o_input_last_unexpected'.
# ** Error: (vopt-7052) ..../top_tb.sv(33): Failed to find 'o_ctrl_data_invalid' in hierarchical name '/o_ctrl_data_invalid'.
# ** Error: (vopt-7052) ..../top_tb.sv(33): Failed to find 'o_ctrl_data_invalid' in hierarchical name '/o_ctrl_data_invalid'.
# Optimization failed
# Error loading design
```

jrwagz, 2019-04-12:
The rs_decoder_7_3_v9_sim_netlist.vhdl file is quite large, but this is how it starts:

```
-- Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.
-- --------------------------------------------------------------------------------
-- Tool Version: Vivado v.2016.4 (lin64) Build 1733598 Wed Dec 14 22:35:42 MST 2016
-- Date        : Fri Feb 15 11:48:35 2019
-- Design      : rs_decoder_7_3_v9
-- Purpose     : This VHDL netlist is a functional simulation representation of the design and should not be modified or
--               synthesized. This netlist cannot be used for SDF annotated simulation.
-- --------------------------------------------------------------------------------
`protect begin_protected
`protect version = 1
`protect encrypt_agent = "XILINX"
`protect encrypt_agent_info = "Xilinx Encryption Tool 2014"
`protect key_keyowner="Cadence Design Systems.", key_keyname="cds_rsa_key", key_method="rsa"
`protect encoding = (enctype="BASE64", line_length=76, bytes=64)
`protect key_block
TRIpJ7TpeIFWnVtwNsgPk4L8SPTxOzutwvlldol8golXBg13D1L3XJUm1+9gf81/hKs5EKC6Upn7
jQLdfEhEgA==

`protect key_keyowner="Mentor Graphics Corporation", key_keyname="MGC-VERIF-SIM-RSA-1", key_method="rsa"
`protect encoding = (enctype="BASE64", line_length=76, bytes=128)
`protect key_block
TC0Kcrcn/9gR/FsVoSRvmZPc3F7pjn/+WYlKOhGrY9+NgmO1T8AwQUZ+G/V0gSEedhDhPnODy7XJ
KFVULn8MfNu7LmqEh6U8iSg4GddkduwH0LecdZyiVGkbKNqFAP2vAqPK2XDdiib2CotmCiAKr8JA
2bWLWPggDdbbQLqSAbM=

`protect key_keyowner="Synopsys", key_keyname="SNPS-VCS-RSA-1", key_method="rsa"
`protect encoding = (enctype="BASE64", line_length=76, bytes=128)
`protect key_block
C7U69bD9WzJZxDuc5n4BeWVxukUyTVxNLBlRkTcQFVEpmSyyL0Bn/rrO+nqo9hIxvRxczMZLMNM6
JBbYdFYBQioPaHBr/BMaY5PH7mcY5N+U4wXcZIP+3/u4c44iEVlPampkfjao9pNUkLmyW7ctXwQM
eRnVTJZoblb7xz8PYhw=

`protect key_keyowner="Aldec", key_keyname="ALDEC15_001", key_method="rsa"
`protect encoding = (enctype="BASE64", line_length=76, bytes=256)
`protect key_block
pAKmfxQwh5VWPl1bDDA+JccsNignBeNVSCjUr6jSi4aQoJySIGMZrhdwsWcI2EiqskWVY7Ttp9CE
9nB9DdulMJYXkcro6TiLwss2b4dvR1g9PDklQcOEc16DZBy8xMUMYsqOS2RbVHB47zzkwQs5VX8Q
WKXOHt+jj26KcBp1WEKNnldgarOhwbK7KMO6f4IIdcAxPPJlPRsqtoq7zZR5fxRwuikgTY/2Kaac
AnXuLLz9RiMx4BKzMDJK+x6W7AQfEPObsVvDuzZ4UZB6Wv6LIPDSnqcZV0gBFjUMNJVjr1x4e0R4
6ZhNSsjfhd18K+7VMBw8J+JskxJui9SKsoPQbA==

`protect key_keyowner="ATRENTA", key_keyname="ATR-SG-2015-RSA-3", key_method="rsa"
`protect encoding = (enctype="BASE64", line_length=76, bytes=256)
`protect key_block
sP4oqzxP1ONXGQyUxRA02jVU9Asb05I8avQm/CBtA2LydFUaH4PsgYKPb8NMnpS0QBqB+DuUuxVB
84txvIeqwXmOuHoMqE2j+dXttY5cJX3dLbAxGEueuyJnk28iYoLox2Lt1Rg6wbvS50n8yNKn+2ea
gUNqNrNtyaXrZCPuF0sW9HLNV2DXsgSSK2INHK9BiT+nWbdDKZny58Tkt8lo3hkq1/tePqYYhcDM
gTpfH3ddKbT7H5FUZjwubEjory20PCjB1TmSehXwB0zFsfjEd9LNcPtMUJSvwG+oQItYwnHPXrsO
4roLo+Kg+Pn+pd1w+Eghvye7oDoKPh8pl8G/jw==

`protect key_keyowner="Xilinx", key_keyname="xilinx_2016_05", key_method="rsa"
`protect encoding = (enctype="BASE64", line_length=76, bytes=256)
`protect key_block
bDIs2jx6PgeMTwADd7ZEUNE9+jNImroXt2OlFF+i4Wy35sZpZeD+ugyeBrJOXBiS8hV2o0I8Ga9i
Baf+izMDUVnaAdQ307OJ3dQx3omEsCjbhrtR2syBKJxcJNiiZPvbrp3Nsn49j578348Bktml/jQJ
SvrgllhQ371hZ0+l8TM/YtjA+sCFyEZk6maE0X/WaGaY5F44qZHKPF/wQtqQd7vajvEW0PKpbINr
6rxSBl37W8mwRLIjbt5A8tXA1n5dqIy4/1O0OoiqQvSxQLhR0bQ+YIqhFDX9mQrVpostUkkaqm8X
TtVVXQ+dZLi8C8R40VQTwVm1ZPhMr3RnkBGasw==

`protect key_keyowner="Mentor Graphics Corporation", key_keyname="MGC-PREC-RSA", key_method="rsa"
`protect encoding = (enctype="base64", line_length=76, bytes=256)
`protect key_block
R8W1pehS+f01L1CCLjZ+i3G4VqRv25ysXqPT+DMW9yeLb3f1LkljM15PLn54IBoxVE1EcMVSOTUG
gZaD0D34XiPLtSv25FQOaYVJk5kZ9CPKbpJMh6x83025DSA08swQY6+j3/q5IILIiGlufO51Hk4i
PSFUsYv+OJKDOUEZ9kBFvFp/PxfOV4fVhTF0DZW3GFytHZaFXtGb0wtihbBeC2fJjY+TljqiCR4d
sz5EIhdq5PY7PnRumuYRRLnhQMW1gLccY/YtZBwEB2DLslkgvQ7QRZCj+8D2JM7RMHy2x0IZ+5Zr
l+Vk6j3xE53ve3rJULY3A5ciizfe7M0cbmUWCQ==
[... 449 lines cut ...]
      aresetn => '0',
      event_s_ctrl_tdata_invalid => NLW_d1_event_s_ctrl_tdata_invalid_UNCONNECTED,
      event_s_input_tlast_missing => event_s_input_tlast_missing,
      event_s_input_tlast_unexpected => event_s_input_tlast_unexpected,
      m_axis_output_tdata(7 downto 0) => m_axis_output_tdata(7 downto 0),
      m_axis_output_tlast => m_axis_output_tlast,
      m_axis_output_tready => m_axis_output_tready,
      m_axis_output_tuser(0) => NLW_d1_m_axis_output_tuser_UNCONNECTED(0),
      m_axis_output_tvalid => m_axis_output_tvalid,
      m_axis_stat_tdata(7 downto 5) => NLW_d1_m_axis_stat_tdata_UNCONNECTED(7 downto 5),
      m_axis_stat_tdata(4 downto 0) => \^m_axis_stat_tdata\(4 downto 0),
      m_axis_stat_tlast => NLW_d1_m_axis_stat_tlast_UNCONNECTED,
      m_axis_stat_tready => m_axis_stat_tready,
      m_axis_stat_tvalid => m_axis_stat_tvalid,
      s_axis_ctrl_tdata(7 downto 0) => B"00000000",
      s_axis_ctrl_tready => NLW_d1_s_axis_ctrl_tready_UNCONNECTED,
      s_axis_ctrl_tvalid => '0',
      s_axis_input_tdata(7 downto 0) => s_axis_input_tdata(7 downto 0),
      s_axis_input_tlast => s_axis_input_tlast,
      s_axis_input_tready => s_axis_input_tready,
      s_axis_input_tuser(0) => '0',
      s_axis_input_tvalid => s_axis_input_tvalid
    );
end STRUCTURE;
```

kraigher, 2019-04-13:
First of all the warning you see is not from modelsim but from the VUnit dependency scanner which cannot find any `rs_decoder_7_3_v9_rs_decoder_v9_0_12_viv` design unit since it is defined by encrypted code. In the VUnit Vivado example we will not use dependency scanning for the encrypted IP (using no_parse option for add_source_file) but rather add a manual dependency.

Have you added unisim libraries etc. to either your simulator built-in libraries or as external libraries in VUnit?

Have you tried commenting out the instantiation of the IP inside the rs_decoder wrapper? The error messages you posted complains about the ports of the rs_decoder wrapper and that should have nothing to do with the IP.

You could try to manually run the common.do file that VUnit creates in the test output path. You could compare the contents of vunit_load with whatever your old simulation command was. Are you able to write manual do-file which is able to simulate this scenario?

jrwagz, 2019-04-15:
I link the unisim, etc. libraries through our modelsim.ini file:

```
setenv VUNIT_MODELSIM_INI ./modelsim.ini
```

And inside the modelsim.ini file looks like this:

```
[Library]
unisim        = /path/to/precompiled/lib/unisim
```

Thanks for the idea to try and build a manual .do file to simulate this.  I found a couple of issues that were masking the problems, and I believe that I've resolved all my issues now.  

First issue, verilog syntax problem in top_tb.sv, I forgot to put the leading period on each line of the ports in the dut instantiation: (face palm)

```
  rs_decoder dut (
    .i_clk                   (clk),
    .o_rd_en                 (),
    .i_dv                    (1'b1),
    .i_eof                   (1'b1),
    .i_data                  (8'hFF),
    .i_rdy                   (1'b1),
    .o_dv                    (),
    .o_eof                   (),
    .o_data                  (out_data),
    .i_stat_ready            (1'b1),
    .o_stat_data             (),
    .o_stat_valid            (),
    .o_input_last_missing    (),
    .o_input_last_unexpected (),
    .o_ctrl_data_invalid     ()
  );
```

And when I was sorting out the details of how I'm referencing the unisim libraries, I found that I was previously mapping the wrong version of the libraries for the version of Vivado that had generated this funcsim model.

After fixing the syntax error and the dependency version problem, the simulations worked without a hitch, since modelsim ended up doing the actual compilation of the funcsim model and was able to find the module that the python dependecy checker couldn't see.

Thanks for the help!

closed 2019-04-15 by jrwagz