VUnit/vunit #477: axi_stream_slave with check function requires TREADY

eschmidscs, 2019-04-16:
Hi

Change 3e098f567af0a442ae9f3a41b0fcb90f2092daed modifies the behaviour of the check_stream/check_axi_stream function.
The check functions call the pop function one by one. The command queue to the VC thus always runs empty, causing TREADY going low with the current implementation. This results in max 50% duty cycle of the checked stream.
However, TREADY is optional by the AXIS spec.

I don't grasp the full background of the change above. Is the only reason to have it implemented the same way?

LG/e

kraigher, 2019-04-16:
A check operation is currently a blocking operation just like a read since the check is done in the client code. A blocking operation will limit the throughput as you notice. To solve the situation you need to queue up many reads using non blocking calls and do the checks afterwards. You can already do this using the non blocking read calls today. 

The situation could be improved in two ways:

- Keep check a client side operation but make a procedure which will work on an entire array such that reads can be queued up internally.
- Move the check into the VC. The client code must add a wait_until_idle call to ensure the check has happened.

eschmidscs, 2019-04-17:
Sidenote: I find the naming confusing here. It is obviously blocking, as it waits for the actual transfer. That it causes the next cycle to be idle is something more than blocking.

The problem was solved before the mentioned commit, so it is not impossible to implement. However, if you want to have the message handler implemented as it is now, it probably does not work (even though being an "artifact" of how the simulator works).

I hope I find time to look at your second option. I think non-blocking checks would be nice anyway.

kraigher, 2019-04-17:
As you mention blocking is only a problem if there is latency from request to response. Maybe the way the AXI stream VC is implemented adds a latency such that the next clock edge is missed. Maybe for AXI stream the latency could be zero and a blocking operation have full throughput. However for a memory mapped AXI port such as AXI4 lite there will typically be a latency and a blocking operation will never have full throughput.

PR #483 (2019-04-24): check_axi_stream non-blocking
This resolves #477 

I reused the same function with an additional argument with default value for backward compatibility (compared to the current master).
I prefer that to a separate function, but let me know if you think otherwise.

Only the specific axi_stream stuff is updated. One could think of doing the same to the generic stream_slave, but I can't handle the consequences (would need to update the uart...).

Running all checks in verification_components seems to fail with questa due to some delta-cycle differences with GHDL... :-/ GHDL looks fine here.
closed 2019-04-24 by kraigher