VUnit/vunit #478: Programatic procedure to retrieve the output path of a test

umarcor, 2019-04-17:
Ref #467.

> @umarcor:
> Is it possible to read `test_name_to_path_mapping.txt`` directly as some Python type (dict or list)?

> @kraigher:
> The file was intended for human use. The format is `folder_name` `SPACE` `test_name`.
I do not recommend that you use it. It is better we add this to `--export-json` instead.

> @kraigher:
> Currently the commands are logged if you use `--log-level=DEBUG` for human use. If an external tool needs this information we need to add it to `--export-json`.

> @umarcor:
> IMHO, there should be a simple and realiable programatic procedure to retrieve the output/log of a test. It don't really mind if I need to set some stdout in the python script, or get it as a return field from `vu.main()`, or parse some output file. I am trying to process `test_name_to_path_mapping.txt` just because I didn't find a better way to 'see' the content in `output.txt`.
>
> About adding it to `--export-json`, although useful, I'd rather avoid it for this specific use case. Using JSON is a better approach when it is to be used from Python, Go, etc. But, with Bash or C I find it easier not to require a JSON parsing library. The current format, as you described it, is straighforward, as one only needs to search for the first space.

kraigher, 2019-04-17:
Are you aware that the output path is accessible as an attribute on the post_check and pre_config hooks as well? Depending on the use case this might be what you need rather than having it written in the export-json file.

umarcor, 2019-04-17:
No, I was not. And I can very probably use it to retrieve the path in the `run.py` file, without actually running the simulation. I will give it a try. Thanks!

kraigher, 2019-04-17:
It will only be accessible when running the pre config hook before simulation. So you need to run the simulation.

umarcor, 2019-04-17:
> It will only be accessible when running the pre config hook before simulation. So you need to run the simulation.

I think I should be able to 'run the simulation' as `--elaborate` with sim_option `ghdl.elab_e`, so the pre config hook is triggered and the binary is compiled (thus, the output path needs to be created), but no simulation is executed. What do you think?

kraigher, 2019-04-17:
That would work.

umarcor, 2019-04-21:
I implemented it in #476 as follows:

``` python
def preconf(output_path):
    global opath
    opath = output_path
    return True

for tb in lib.get_test_benches(pattern='*', allow_empty=False):
    tb.set_pre_config(preconf)

ui.set_sim_option("ghdl.elab_e", True)
ui._args.elaborate = True
try:
    ui.main()
except SystemExit as exc:
    if exc.code is not 0:
        exit(exc.code)

args = [line.rstrip('\n') for line in open(join(opath, 'ghdl', 'args.txt'))]
```

So, it would be useful if `ui.main()` wrote the args of each test/config to a global variable of type 'dict of lists', without requiring 'pre_config' or 'post_check' hooks. Something similar to:

``` python
ui.set_sim_option("ghdl.elab_e", True)
ui._args.elaborate = True
try:
    ui.main()
except SystemExit as exc:
    if exc.code is not 0:
        exit(exc.code)

for i, x in enumerate(args):
    print(i) # entity.arch.config
    print(x) # list of args (argv to GHDL)
```

Alternatively, it can be implemented in `--export-json`, as @kraigher suggested. @kraigher, what fo you think? Should be add all the runtime args (those after the name of the arch) to `--export-json`? Note that the first arg is the path of the binary itself; thus, the output path is provided implicitly.

PR #578 (2019-11-16): post_run: add report data to 'Results' object/class
Close #445
Close #478
Close #574

In this PR, parameters `output_path` and `report` are passed from `main` to the `Results` object built and passed to `post_run` callbacks. Instead of exposing the `TestReport` object, as suggested in #445, a subset of info is extracted for each test: name, status, time and path. The path is relative to `join(output_path, 'test_output')`.

Usage example:

```py
def post_func(results):
    print(results.output_path)
    for key, val in results.get_report().items():
        print(key , val)

vu.main(post_run=pos [... cut]
closed 2019-11-16 by 1138-4EB