VUnit/vunit #479: Exporting Simulation for Questa

dsp20, 2019-04-17:
I have Created a simulation project in vivado and then exporting that simulation for Questa. To run this simulation using Vunit environment I have created python script. Exported Simulation is creating a separate library folder for each IP used in a project created in vivado. In python script, I have to give a path to each IP library folder otherwise compilation giving an error, though I have compiled all Xilinx IPs for Questa. Each time I create new project I have to modify python script to specify library path for xilinx IP created in export simulation folder. Is there any way we can give only common path for any xilinx IP compilation?

kraigher, 2019-04-17:
I need more information to be able to help you. Could you give me an example or be more specific in your problem description?

By the way have you looked at the VUnit Vivado [example](https://github.com/VUnit/vunit/tree/master/examples/vhdl/vivado). It provides an automatic way to compile and add Vivado IP into a run.py file. It is the approach I would recommend and that I am using myself at work where I also use Vivado.

dsp20, 2019-04-18:
So, I have created a vivado project and it has AXI VIP and two Video test pattern generator IP and AXI interconnect. Simulation works fine in Vivado simulator. To run the simulation in Questa sim I am exporting simulation with source files. Now exported simulation has folders "incl", "IP". I have written a python script for Vunit environment. Now "IP" folder has supported library files of IPs which I used in the project. So here I have to mention path for each IP related files in python script. Otherwise, compilation gives an error. As these IPs are from Vivado I expect It should get compiled just by giving a common library path. Otherwise every time I create a new project with Vunit Environment I have to mention specific IPs path from the exported simulation. modelisim.ini file has all the compiled library file location for xilixn IPs, still, I get an error if I do not give a specific path for each IP library files

kraigher, 2019-04-23:
Could you give me an example or show some code how you imagine this would work? I do not really understand what it is you want.

Also remember a run.py file is just a Python script and Python is a full featured programming language so you can create any abstraction you want on top of VUnit if you find some common pattern in your run.py files.

dsp20, 2019-04-25:
[image]
For Xilinx IPs, I need to give the path for the IP library from a python script. As I am using Questasim, I have compiled all libraries using "complile_simlib" command from vivado and stored to the folder. Now in Vunit environment, I need to add library file for those IPs.  But compiled library folder has files like above image. so compilation is not working.

dsp20, 2019-04-25:
For simulation, I am getting the error shown in the image. The compiler is not able to find unisim , unisim.vcomponents libraries. It is also giving error for Xilinx IPs.

kraigher, 2019-04-26:
I do not really know how I can help you. My general advice would be to start with the Vivado example project from VUnit and check that it even works in your environment. Step 2 would be to try to adapt it to your project one step at a time and paying attention when it stops working.

dsp20, 2019-05-02:
Yes. I started with vivado Example project and after running run.py script Xilinx IPs are getting compiled and after IPs compilation, execution stops with the error 

"compilation failed due to previous error"

.  At the start of the compilation, I am getting an error

>  "Failed to find the 'sccom' executable path! Trying standard GNU compiler from current installation environment."

 and after that, all ips are getting compiled and execution stops. Related to "sccom" error Xilinx says we can ignore it and they will rectify that in a future version.