VUnit/vunit #484: Reconfiguring Test Cases

dsp20, 2019-04-23:
Is it possible to reconfigure the number of a test case? For example, with the help of a single parameter number of test cases will be generated. For parameter 3, three test cases will be generated and if I change the value to 5, five test cases will be generated. Is this kind of reconfiguration is possible?

kraigher, 2019-04-23:
Yes you can use the `add_config` method to add many configurations for a single test case such as running it 10 times with different generic settings. There are examples for this in the documentation on the website.

dsp20, 2019-04-24:
I mean I want to Number of Test cases to be configurable. For example, for one run I want 4 test cases and for another run, I want 6 test cases by changing the signle parameter from a python script.  Is it possible?

umarcor, 2019-04-24:
> I mean I want to Number of Test cases to be configurable. For example, for one run I want 4 test cases and for another run, I want 6 test cases by changing the signle parameter from a python script. Is it possible?

Yes, you can define a custom arg that you pass to the python script, and then add as many configurations as you want. See http://vunit.github.io/python_interface.html?highlight=custom#vunit.ui.VUnit.from_args

kraigher, 2019-04-26:
Question was answered so we can close

closed 2019-04-26 by kraigher