VUnit/vunit #485: Running multiple vunit test  in separate processes failing

mickey4u, 2019-04-23:
I have a c++ program that is suppose to run vunit tests in separate process as shown in the c++ code snippet below but I get the following errors logs. I have no idea what this means.

```
int p1 = fork();
    int p2 = fork();
    int p3 = fork();
    int p4 = fork();

    if (p1 == 0) {
        string cd = " cd Exp/process1 && ";
        string pyth = " python run.py --unique-sim &";
        string COMMAND = cd + pyth;
        system(COMMAND.c_str());

    } else if (p2 == 0) {
        string cd = " cd Exp/process2 && ";
        string pyth = " python run.py --unique-sim &";
        string COMMAND = cd + pyth;
        system(COMMAND.c_str());

    } else if (p3 == 0) {
        string cd = " cd Exp/process3 && ";
        string pyth = " python run.py --unique-sim &";
        string COMMAND = cd + pyth;
        system(COMMAND.c_str());

    } else if (p4 == 0) {
        string cd = " cd Exp/process4 && ";
        string pyth = " python run.py --unique-sim &";
        string COMMAND = cd + pyth;
        system(COMMAND.c_str());

    }
```

Console log

```
Traceback (most recent call last):
  File "run.py", line 20, in <module>
    lib.add_source_files(join(root, absolutePath, "*.sv"))
  File "/home/mydir/.local/lib/python2.7/site-packages/vunit/ui.py", line 1274, in add_source_files
    for file_name in file_names])
  File "/home/mydir/.local/lib/python2.7/site-packages/vunit/ui.py", line 1310, in add_source_file
    self._library_name, file_name, preprocessors)
  File "/home/mydir/.local/lib/python2.7/site-packages/vunit/ui.py", line 743, in _preprocess
    ostools.write_file(pp_file_name, code, encoding=HDL_FILE_ENCODING)
  File "/home/mydir/.local/lib/python2.7/site-packages/vunit/ostools.py", line 306, in write_file
    os.makedirs(path)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 17] File exists: '/home/mydir/CLionProjects/LearnThreads/Exp/process2/vunit_out/preprocessed/lib'
Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/1_mux.sv                                       Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/1_mux.sv                                       Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         Compiling into lib:       vunit_out/preprocessed/lib/1_mux.sv                                       Compiling into lib:       vunit_out/preprocessed/lib/mux.sv                                         passed
Compiling into vunit_lib: ../../../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv passed
Compiling into vunit_lib: ../../../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv passed
Compiling into vunit_lib: ../../../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv passed
Compiling into vunit_lib: ../../../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv failed
=== Command used: ===
/opt/intelFPGA_lite/18.1/modelsim_ase/bin/vlog -quiet -modelsimini /home/mydir/CLionProjects/LearnThreads/Exp/process1/vunit_out/modelsim/modelsim.ini -sv -work lib /home/mydir/CLionProjects/LearnThreads/Exp/process1/vunit_out/preprocessed/lib/1_mux.sv -L vunit_lib -L lib +incdir+/home/mydir/.local/lib/python2.7/site-packages/vunit/verilog/include

=== Command output: ===
** Error: (vlog-7) Failed to open design unit file "/home/mydir/CLionProjects/LearnThreads/Exp/process1/vunit_out/preprocessed/lib/1_mux.sv" in read mode.
No such file or directory. (errno = ENOENT)

Compile failed
passed
Compiling into vunit_lib: ../../../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv passed
Compiling into vunit_lib: ../../../../.local/lib/python2.7/site-packages/vunit/verilog/vunit_pkg.sv failed
=== Command used: ===
/opt/intelFPGA_lite/18.1/modelsim_ase/bin/vlog -quiet -modelsimini /home/mydir/CLionProjects/LearnThreads/Exp/process1/vunit_out/modelsim/modelsim.ini -sv -work lib /home/mydir/CLionProjects/LearnThreads/Exp/process1/vunit_out/preprocessed/lib/1_mux.sv -L vunit_lib -L lib +incdir+/home/mydir/.local/lib/python2.7/site-packages/vunit/verilog/include
[... 200 lines cut ...]
Output file: vunit_out/test_output/lib.tb_mux.Testing_mux_7122379a18d17e6080943173657079cc5b07344f/output.txt
fail (P=0 S=0 F=1 T=1) lib.tb_mux.Testing mux (1.0 seconds)

==== Summary ==================================
fail lib.tb_mux.Testing mux (1.0 seconds)
===============================================
pass 0 of 1
fail 1 of 1
===============================================
Total time was 1.0 seconds
Elapsed time was 1.0 seconds
===============================================
Some failed!
pass (P=1 S=0 F=0 T=1) lib.tb_mux.Testing mux (0.8 seconds)

==== Summary ==================================
pass lib.tb_mux.Testing mux (0.8 seconds)
===============================================
pass 1 of 1
===============================================
Total time was 0.8 seconds
Elapsed time was 0.8 seconds
===============================================
All passed!
```

kraigher, 2019-04-23:
```c
    int p1 = fork();
    int p2 = fork();
    int p3 = fork();
    int p4 = fork();
```

This will create 2**4 = 16 processes. I think you only want to create 4. This will cause the same run.py file to execute in the same folder such as process2 and it will cause all kind of race conditions.

mickey4u, 2019-04-23:
@kraigher Thank you. Yes, I want to create only 4 processes to run in parallel. And please how exactly does it end up creating 16 processes and how can I fix it. Thanks.

kraigher, 2019-04-23:
```c

    int p1 = fork();
    // Now there are 2 processes
    // 1. p1 = 0
    // 2. p1 = 1
    int p2 = fork();
    // Now there are 4 processes
    // 1. p2 = 0, p1 = 0
    // 2. p2 = 1, p1 = 0
    // 3. p2 = 0, p1 = 1
    // 4. p2 = 1, p1 = 1
    int p3 = fork();
    // Now there are 8 processes
    // ...
    int p4 = fork();
    // Now there are 16 processes
    // ...
```

Use if-statements around subsequent forks such that p2 is only created if p1=0 etc.

umarcor, 2019-04-23:
@mickey4u, just because of curiosity, do any of the processes/tests communicate with each other? Or are you running them in parallel to reduce execution time?

mickey4u, 2019-04-23:
@umarcor No there is no communication between these tests. Parallel execution to reduce execution time is what I want to achieve.

mickey4u, 2019-04-24:
@kraigher Is a new simulator instance called everytime `python run.py` is used? If so, how do I use the same simulator instance for testing say 2k plus design modules? Because I'm guessing starting a new instance consumes most of the evaluation time.

kraigher, 2019-04-24:
@mickey4u Yes a new simulator instance is spawned by each run.py invocation. When not running with `--unique-sim` a simulator instance is shared during the execution of a run.py file. Regarding running simulations in parallel it is already supported by VUnit and should not require an external program like you have made.

mickey4u, 2019-04-24:
@kraigher I thought the parallel option was for different testcases within the same testbench against a design module. My specific use case is, I have a number of design modules `adder1.sv adder2.sv .... adder100.sv ` and a single testbench `tb_adder.sv` which should be used to test all these design modules.

kraigher, 2019-04-24:
Your problem is then not about parallel execution but having multiple modules with the same name coexist within the same project. You could solve this problem by either having one design library per module or by generating code where the instance name is filled into a template.

mickey4u, 2019-04-24:
@kraigher Please, how do i achieve the instance name getting filled into a template.

mickey4u, 2019-04-25:
@kraigher Did you mean something like the code snippet below
```
from vunit.verilog import VUnit
from os.path import dirname, join
from os import path

root = dirname(__file__)
for module in [list of modules]:
    vu = VUnit.from_argv()
    vu.enable_check_preprocessing()
    lib = vu.add_library("lib", True)
    absolutePath = path.abspath("")
    lib.add_source_files(join(root, absolutePath, "*.sv"))
   vu.main()
```
and using a command like `python run.py -p6` will use parallel execution.

umarcor, 2019-04-25:
@mickey4u, that will generate `len([list of modules])` projects, one each time `vu = VUnit.from_argv()` is executed. In order to define multiple libs for a single project, you need to create `vu` outside of the for loop.

mickey4u, 2019-04-25:
from vunit.verilog import VUnit
from os.path import dirname, join
from os import path

root = dirname(__file__)

@umarcor Something like this

```
vu = VUnit.from_argv()
for module in [list of modules]:
    lib = vu.add_library("generate unique lib names", True)
    lib.add_source_files(join(src_path, "test", "*.sv"))
    lib.add_source_files(join(src_path, "test", "modules", module))
vu.main()
```

umarcor, 2019-04-25:
That's it. Let's say that all of you modules have the same testbench entity (tb_test) and the same test name (test). If you generate libs `libA`, `libB`, etc. Then you will have the following list of available tests in the project:

- `libA.tb_test.test`
- `libB.tb_test.test`
- `libC.tb_test.test`

You can check if the project is being properly built by executing `python3 run.py -l`, which will only list them.

mickey4u, 2019-04-25:
@umarcor Thanks.

umarcor, 2019-04-26:
Note that, alternatively, you can write a single testbench where all of the modules are instantiated, and a test is defined for each of them:

``` vhdl

signal en: std_logic_vector(2 downto 0);

main: process begin
  test_runner_setup(runner, runner_cfg);
  while test_suite loop
    if run("test") then
      en <= "001";
      ...
    elsif run("testB") then
      en <= "010";
      ...
    elsif run("testC") then
      en <= "100";
      ...
    end if;
  end loop;
  test_runner_cleanup(runner);
  wait;
end process;
...
-- Some common resources which are used indendently of the selected case
...
uutA: entity libA.entity.arch
  port map ( en => en(0) );
...
uutB: entity libB.entity.arch
  port map ( en => en(1) );
...
uutC: entity libC.entity.arch
  port map ( en => en(2) );
...
```

Which will produce the following list:

- `lib.tb_test.testA`
- `lib.tb_test.testB`
- `lib.tb_test.testC`

mickey4u, 2019-04-26:
@umarcor the only problem is the number of design modules are a lot, 1000+.

umarcor, 2019-04-26:
I see. Nevertheless, you are using python to create the project. You can do code generation online:

- Write some template of the main testbench and leave two placeholders (one in the while loop where test names are defined, and a second one where entities are to instantiated).
- Then, from python, just generate some dict/list/struct with 1000+ entries and write a simple generation function to expand all of them in the template.
- Apply the dict to the template.
- Profit.

I have successfully done this with golang templates (https://gowebexamples.com/templates/). But I am pretty sure that a similar feature for python must exist.

kraigher, 2019-04-26:
I would say it is probably easier to generate separate test benches for each module rather than one big test bench. It would also be easier to debug.

mickey4u, 2019-04-26:
@kraigher Okay

mickey4u, 2019-04-28:
@umarcor @kraigher For some unknown reason, the testbenches are writing the same value to  their respective files but  when I test them one after the other they write different values. What could I have done wrong?

```
for module in ["dut1.sv", "dut2.sv"]:
    # create the library 'lib'
    lib = vu.add_library(module[:6], True)
    lib.add_source_files(join(root, absolutePath, module))
    lib.add_source_files(join(root, absolutePath, "test", "tb_"+ module))
# run vunit function
vu.main()
```

kraigher, 2019-04-29:
> For some unknown reason, the testbenches are writing the same value to their respective files but when I test them one after the other they write different values. 

I do not get a clear picture of what you mean by this. Could you elaborate and show an example with more details.

mickey4u, 2019-04-29:
@kraigher So basically I created two modules `muxA.sv` and `muxB.sv` and their respective testbenches `tb_muxA.sv` and `tb_muxB.sv`. These testbenches write to separate files the number of test cases passed by the modules. What I am observing is the number of testcases passed by one of the modules is written into both files. But when I test them individually they output the different values into the files.

kraigher, 2019-04-29:
Do you create a unique file name for each test bench?

mickey4u, 2019-04-29:
yes the files are unique.

kraigher, 2019-04-29:
Ok then you have some other bug in your code

mickey4u, 2019-04-29:
Testbench: the files they write to differs
```
`include "vunit_defines.svh"

module tb_mux;

    bit a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7, out;

    // read
    bit expectedOutput;

    int passedFile;
    integer i;

    bit [11:0] testvector[2047:0];

    // design unit under test instantiation
    mux dut(.out(out), .a0(a0), .a1(a1), .a2(a2), .d0(d0), .d1(d1), .d2(d2), .d3(d3), .d4(d4), .d5(d5), .d6(d6), .d7(d7));

    `TEST_SUITE begin

        `TEST_SUITE_SETUP begin
            $readmemb("../../testcases", testvector);

            // open file writing
            passedFile = $fopen("../../pass.txt", "w");
            if(passedFile)
                $display("file opened");
            else
                $display("couldn't open the file");
        end

        `TEST_CASE("Testing mux") begin
            $display("Testing started");
            assertOutput();
        end

        `TEST_CASE_CLEANUP begin
            $fclose(passedFile);
        end
    end;

task automatic assertOutput();
     automatic int unsigned fitness = 0;
    for (i = 0; i< 2048; i++)begin
        {a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7, expectedOutput} = testvector[i];
        //$fdisplay(passedFile, "%b %b %b %b %b %b %b %b %b %b", a, d0, d1, d2, d3, d4, d5, d6, d7, expectedOutput);

       #10;
           if(a0 == 1'b0 && a1 == 1'b0 && a2 == 1'b0 && out == expectedOutput)
                fitness = fitness + 1;

           else if(a0 == 1'b0 && a1 == 1'b0 && a2 == 1'b1 && out == expectedOutput)
               fitness = fitness + 1;

           else if(a0 == 1'b0 && a1 == 1'b1 && a2 == 1'b0 && out == expectedOutput)
               fitness = fitness + 1;

           else if(a0 == 1'b0 && a1 == 1'b1 && a2 == 1'b1 && out == expectedOutput)
               fitness = fitness + 1;

           else if(a0 == 1'b1 && a1 == 1'b0 && a2 == 1'b0 && out == expectedOutput)
               fitness = fitness + 1;

           else if(a0 == 1'b1 && a1 == 1'b0 && a2 == 1'b1 && out == expectedOutput)
               fitness = fitness + 1;

           else if(a0 == 1'b1 && a1 == 1'b1 && a2 == 1'b0 && out == expectedOutput)
               fitness = fitness + 1;

           else if(a0 == 1'b1 && a1 == 1'b1 && a2 == 1'b1 && out == expectedOutput)
               fitness = fitness + 1;

           else  $display(0);
    end
    $fdisplay(passedFile, "%d", fitness);

endtask : assertOutput;

endmodule
```

midimaster21b, 2019-09-24:
Can this issue be closed?

1138-4EB, 2019-09-24:
> Can this issue be closed?

I think so!

closed 2019-09-24 by 1138-4EB