VUnit/vunit #486: I am

dsp20, 2019-04-24:
I am referring this example for compiling Xilinx IPs from Vunit environment. But I am getting simulator related error. I am using QuestaSim 10.6d and also added PATH variable for this. Please find attached file for error. Do I need to change anything in compile_standard_libs?tcl file?
[image]

kraigher, 2019-04-24:
It looks like Vivado fails to run vcom.exe. Maybe something with 32-bit vs 64-bit mode? You could try to update the tcl script to use 64-bit instead.

LarsAsplund, 2019-05-29:
@dsp20 Any progress? Should the issue remain open or can it be closed?

dsp20, 2019-05-30:
Now I am not compiling Xilinx IP using Vunit environment.

LarsAsplund, 2019-05-31:
@dsp20 Ok, in that case I'll close this issue

closed 2019-05-31 by LarsAsplund