VUnit/vunit #489: Unism Library not found

dsp20, 2019-04-26:
I am getting unisim library related error. I have compiled Xilinx IPs for Questasim. Still, compilation giving an error. What could be the solution for this?
[image]

kraigher, 2019-04-26:
The only reason I can think of is that the unisim library is not compiled and added to the project.

dsp20, 2019-05-02:
I have modified modelsim.ini file and now able to see all library files in Questasim. Please see attached image 
[image]
In this image we can see "proc_sys_reset_v5_0_13" library is added in Questasim. But while compiling  project from Vunit it is giving an error for "proc_sys_reset_v5_0_13". Please see attached command prompt image
[image]

Do I need to add any script in run.py related to proc_sys_reset_v5_0_13 library?

LarsAsplund, 2019-05-31:
@dsp20 Can this also be closed?

dsp20, 2019-05-31:
Yes. This can be closed

closed 2019-05-31 by LarsAsplund