VUnit/vunit #491: Configuration Parameter in System Verilog Testbench

dsp20, 2019-05-10:
How can we set configurable parameters in system verilog testbench?. Like in VHDL tenstbench we use **runner_cfg: string** and in **test_runner_setup**.
Also for each test case entire simulation resets and start again. What if I want to check multiple test cases in a single run only? What could be the requirement for this in system verilog testbench.

LarsAsplund, 2019-05-10:
An example for a SV testbench can be found [here](http://vunit.github.io/user_guide.html#systemverilog-test-benches). Info on setting parameters is in the interface [documentation](http://vunit.github.io/python_interface.html)

dsp20, 2019-05-15:
The example you shared does not show parameter passing. I want to pass parameters from **python script to** **system Verilog testbench**

LarsAsplund, 2019-05-16:
What you're looking for is the set_parameter method in the API documentation.

dsp20, 2019-05-16:
Yes, that is for python side. What should we add in system Verilog testbench side? Like we add runner_cfg: string for VHDL testbench.

LarsAsplund, 2019-05-17:
Here is one example: https://www.edaplayground.com/x/2NJb

LarsAsplund, 2019-05-21:
@dsp20 Have you managed to solve your issue?

dsp20, 2019-05-22:
Yes. I referred files from [](https://github.com/VUnit/vunit/tree/master/vunit/test/acceptance/artificial/verilog) this link. Now able to pass the parameter values from python to Verilog testbench

closed 2019-05-22 by LarsAsplund