VUnit/vunit #496: Attribute error: 'NoneType' object has no attribute 'write'

petandpal, 2019-05-21:
Hello,

I am a new user of vunit and was trying to run the 'VHDL user guide' example provided with Questasim and Python 3.4/2.7.

However I get the above error after successful compilation.  It seems like no tests are being found based on the nonetype object being used. 

[image]

How do I resolve this issue?

1138-4EB, 2019-05-21:
I think that the test is being found because `Starting lib.tb_example.all` is found. You can check it with `python3 run.py -l`. The error seems to be produced when it tries to print/flush some content (expectedly the output of the simulation, either to stdout or output.txt), and for some reason it is empty.

I would suggest to provide us the full sequence of commands that you are introducing.

petandpal, 2019-05-21:
Here is my sequence of commands:

Downloaded tb_example.vhd, tb_example_many.vhd and run.py into working directory.

Linux prompt> setenv PYTHONPATH /apps/elec/ic/vunit/vunit_hdl-4.0.8
Linux prompt> python run.py

Could this be a problem with Colorama? I had to comment out a portion of the 'wrap' definition in test_runner.py because colorama could not be found as shown below:

<<<<<<<<<<<
def wrap(file_obj, use_color=True):
    """
    Wrap file_obj in another stream which handles ANSI color codes using colorama

    NOTE:
    imports colorama here to avoid dependency from setup.py importing VUnit before colorama is installed
    """
    """
    from colorama import AnsiToWin32

    if use_color:
        return AnsiToWin32(file_obj).stream

    return AnsiToWin32(file_obj, strip=True, convert=False).stream
    """
>>>>>>>>>>>>

1138-4EB, 2019-05-21:
> Downloaded tb_example.vhd, tb_example_many.vhd and run.py into working directory.
> 
> Linux prompt> setenv PYTHONPATH /apps/elec/ic/vunit/vunit_hdl-4.0.8
> Linux prompt> python run.py

- Is `python` Python 2 or Python 3? I suggest using Python 3.
- How did you install VUnit?
- Do you have any other supported simulator installed on the system? If so, you should set an envvar to select one. Nonetheless, I think that this is not a problem for you, because VUnit explicitly complains about it when it cannot find a simulator.

> Could this be a problem with Colorama? I had to comment out a portion of the 'wrap' definition in test_runner.py because colorama could not be found as shown below:

I think this is not a good approach. I suggest to install colorama with `pip3 install colorama`.

> Yes. When I run python run.py -l, it finds 3 tests - lib.tb_example.all, lib.tb_example_many.test_pass, lib.tb_example_many.test_fail.

This means that sources are properly parsed and the tests are found. Before investigating further, let's see if installing colorama and reverting the changes you did can fix the issue.

petandpal, 2019-05-22:
So I had Colorama installed, uncommented out my changes to the vunit test_runner.py and re-ran with Python3. I got the error that Colorama could not be found. But when I ran it with Python2, it worked and the test ran. So I will stick with Python2 for now.

1138-4EB, 2019-05-22:
> So I had Colorama installed, uncommented out my changes to the vunit test_runner.py and re-ran with Python3. I got the error that Colorama could not be found. But when I ran it with Python2, it worked and the test ran. So I will stick with Python2 for now.

Then, it is possible you installed colorama with `pip`, instead of `pip3`. That's why it is not found.

Note that Python2 will be deprecated by the end of this year: https://pythonclock.org/

kraigher, 2019-05-26:
The question was answered by @1138-4EB. Install dependencies instead of commenting out code and it will work.

closed 2019-05-26 by kraigher