VUnit/vunit #497: vunit work library issue

jezbaker110, 2019-05-21:
I have an issue where I have tried to list all my project files in the python script. However the code has some IP files which reference packages in a work library. I cannot change the IP code. Vunit is saying it cannot find these files. Vunit will not let me compile these files into a work library, so I am not sure how to get around this issue. It not allowing the design to be compiled. I have checked the modelsim.ini file and no reference has been made to the work library.

1138-4EB, 2019-05-21:
The work library is a special name in VHDL which means 'the same library where this element is defined'. If you compile a module A to library 'lib' and it uses `work.B`, then module B must also be compiled to library 'lib'.

kraigher, 2019-05-26:
The question was answered by @1138-4EB

closed 2019-05-26 by kraigher

jezbaker110, 2019-05-26:
I have tried compiling these pkg files into another library but then when vunit compiles the code,  it throws up an error because it cannot find these files in the work library being referenced.

1138-4EB, 2019-05-26:
Almost all the [examples](https://github.com/VUnit/vunit/tree/master/examples/vhdl) instantiate components from `work`. You can have a look at them as a reference. Apart from that, it is really hard to guess a solution, if you don't help us by providing some detailed info about your use case. Can you provide a MWE? Or, at least, the `run.py` that you are using and the error message you get?

jezbaker110, 2019-05-26:
I was just using the normal method in my run.py file

from os.path import join, dirname
from vunit import VUnit
root = dirname(__file__)
ui = VUnit.from_argv()
lib = ui.add_library("lib")
lib.add_source_files(join(root, "*.vhd"))

The error was telling me that it could not find the library work.
This was when it compiled a file was referencing a pkg file in the library work.

My modelsim.ini file made no reference to the work library.

1138-4EB, 2019-05-26:
That sounds as a problem with the compilation order. Since you are compiling all the sources to the same library, all the components must be found unless i) it was not compiled before or ii) some error was produced while it was being compiled. Please, check the log to ensure that the offending module has not been compiled. If so, rewrite the `run.py` to add sources explicitly in an order which you know that will work. If this solves your issue, please open a new PR about why the order is not properly picked for you design.

jezbaker110, 2019-05-26:
Thanks for the reply. I thought the same thing and tried to compile the pkg files first, using the order that I know works when just compiling into modelsim, but I still had issues.

LarsAsplund, 2019-05-26:
@jezbaker110 We need to see some logs and code to support you on this.