VUnit/vunit #498: Tests that log heavily are killed due to memory starvation

umarcor, 2019-05-21:
The following example will progressively allocates memory until it takes all the available RAM and swap; it is killed by the kernel. However, if the same binary is executed directly (i.e., without Python), the memory starvation error is not present. The issue is independent of `-v`.

I think this might be caused by VUnit buffering all the output from the simulator until the test is finished. A possible enhancement would be to stream it to `output.txt` as I think that it is done for stdout.

I run tests with GHDL (both with mcode and LLVM backends).

``` vhdl
entity tb_memleak is
  generic ( runner_cfg : string );
end entity;

architecture tb of tb_memleak is begin
  process begin
    for x in 0 to integer'high/16 loop
      report "This is some content to be printed";
    end loop;
    wait;
  end process;
end;
```

``` python
from os.path import join, dirname
from vunit import VUnit

vu = VUnit.from_argv()
vu.add_library("lib").add_source_files(join(dirname(__file__), "*.vhd"))

vu.main()
```

kraigher, 2019-05-26:
The test runner does not buffer stdout in memory but uses the file. The problem is when the XML report is to be created, the library we use needs create the file in memory. I would be a useful enchanement to  investigate if the XML-file can be created in a streaming way.

umarcor, 2019-06-20:
> The test runner does not buffer stdout in memory but uses the file. The problem is when the XML report is to be created, the library we use needs create the file in memory. I would be a useful enchanement to investigate if the XML-file can be created in a streaming way.

Is the XML file generated even when `--xunit-xml` is not used?