VUnit/vunit #499: Feature Request: documentation walking through setting up CI with VUnit

GlenNicholls, 2019-05-22:
I am working on setting up TravisCI and possibly exploring the option of other CI environments as well. However, I am completely new to this and do not know how to accomplish this for an open source VHDL code base I am working on.

Is it possible to add instructions to the documentation for setting up a simple example project with TravisCI and GitHub? 

When considering using a different simulator other than GHDL, is a user's docker/containter used instead to avoid installing ModelSim/QuestaSim/Active-HDL/etc. each time a commit is made?

1138-4EB, 2019-05-22:
Hi @GlenNicholls! First, I would suggest to have a look at http://vunit.github.io/cli.html#continuous-integration-ci-environment. If you have used docker before, you will probably know all of it already, but I think it is worth a quick read.

Then, you can see https://github.com/ghdl/docker/blob/master/USE_CASES.md. It explains how to use the GHDL docker images that we provide along with VUnit.

> However, I am completely new to this and do not know how to accomplish this for an open source VHDL code base I am working on.

I am not really sure about what you mean with 'completely new'. Could you please elaborate so it is easier for us to help you? I will assume that you feel comfortable with VUnit, GHDL and/or ModelSim/QuestaSim; that you already know what Docker is and how it is used; and that you only need help with CI services.

> Is it possible to add instructions to the documentation for setting up a simple example project with TravisCI and GitHub?

There are multiple examples about this, but different projects have applied completely different approaches:

- VUnit can be retrieved as a submodule of your repo, as an independent git repo, through pip3, as a tarball/zip, etc.
- The simulator can be installed natively on the Travis VM, or inside a docker container.
- (custom) Docker images can be built as part of the tests, or they can be generated as a completely independent project and just consumed in yours.
- VUnit can be installed in the same environment as the simulator, but it is also possible to have the simulator installed in a docker container and run VUnit from the host.

For example, VUnit itself (this repo) uses TravisCI, GitHub and GHDL to run all the acceptance tests. However, GHDL is compiled from sources each time: https://github.com/VUnit/vunit/blob/master/.travis.yml#L67-L74.

Instead, I would strongly suggest to start with something like:

``` yaml
os: linux
services: docker
language: minimal
install: skip

script: docker run --rm -tv $(pwd):/src -w /src ghdl/ext:vunit bash -c "python3 run.py -v"
```

Assuming that your project has a `run.py` script in the root of the repo, this `.travis.yml` will execute it in a container where GHDL and VUnit are already available.

> When considering using a different simulator other than GHDL, is a user's docker/containter used instead to avoid installing ModelSim/QuestaSim/Active-HDL/etc. each time a commit is made?

Once again, there are different approaches. Yes, it is common for users/companies to have EDA tools such as ModelSim/QuestaSim/Active-HDL/Vivado distributed as docker containers.  Syntactically, you could just replace `ghdl/ext:vunit` with any other docker image (say `myedatools/questasimwithvunit`) and it will work.

But this is completely out of the scope of Travis because of licensing issues. The following requirements must be fulfilled:

- You cannot host the docker image in any public registry (such as hub.docker.com), because you are not allowed to distribute closed-source EDA tools. Hence, you need you CI service (Travis) to be able to communicate with a private registry of yours.
- You must manage a license server and a license file.
  - You can run your own independent server (at home, or at work). Then, you need to allow the container inside the VM in Travis to communicate with it. The risk here is other users being able to also use your licenses.
  - You can start a different license server inside each container. Containers allow to set the MAC address explicitly, so all the 'security' provided by FLEXlm (which binds a license file to a MAC address) is avoided. However, I don't know if this is allowed by the license, I do not think so. You can argue that you are doing a fair use of the license by never running more containers at the same time than you are allowed to. But it is risky, at least. Furthermore, if you put the license file inside the container, any user which has the container can run as many instances as he/she wants with your license file.

That's why the acceptance tests of VUnit in Travis are only executed with GHDL.

> I am working on setting up TravisCI and possibly exploring the option of other CI environments as well. 

If you have never used a CI service, Travis is ok for a first approach, since you do not need to handle credentials or tokens. However, in the long term, I suggest you to check other alternatives, such as CircleCI, GitLab, Bitbucket, GitHub Actions, etc.

This is because Travis was split in `.org` (open source public projects) and `.com` (paid private projects). They've been a year trying to migrate every project to `.com`, but there seems to be little progress. As a result, documentation and UX is quite confusing when you try to run complex setups.

Furthermore, there are frequent communication issues when retrieving dependencies from GitHub, because it is such a used service that API rate limits arise. For example, one of the tests in https://travis-ci.com/ghdl/docker/builds/112705911 has been failing for the last 2 hours, because it cannot retrieve a tarball from GitHub in the deployment stage.

LarsAsplund, 2019-05-22:
@GlenNicholls I'm a bit reluctant to add such information to our documentation as I fear it would require a lot of support and maintenance. CI is a big topic and most stuff has nothing to do with VUnit and our time is limited so we need to focus on the "core".

However, I recognize that this is an important topic so a compromise could be to setup a wiki page with user provided examples and external references to other open source projects, blogs, articles etc using VUnit. This would be less formal than our documentation but more organized than the type of info you find in our Gitter channel. It will also give us more insight in how VUnit is used. It will of course require that people are willing and allowed to share such information about their projects.

LarsAsplund, 2019-05-23:
@GlenNicholls @1138-4EB Here is a very lazy start for you guys to improve upon https://github.com/VUnit/vunit/wiki/CI-Tool-Integration. Ok?

closed 2019-05-26 by LarsAsplund