VUnit/vunit #500: when run a regression of tests without -q, ctrl+c can not abort the execution

zhajio1988, 2019-05-22:
hi, everyone, i write a script for running regression test with some features (https://github.com/zhajio1988/YASA) which use testrunner and reporter from vunit。
but i have encounter some problems。
[image]
when running regression without -q option。that is to say print result to stdout。 when use ctrl+c
to abort the execution. it need multi hit of ctrl+c keyborad, and the final report will not print to stdout.
but when running with -q option, everying is ok.
any one can help me about this issue. thanks.

LarsAsplund, 2019-05-22:
@zhajio1988 I'll leave the original question to @kraigher but I have a few comments on your [YASA project](https://github.com/zhajio1988/YASA).

It looks like you're creating a test framework more targeting SystemVerilog/UVM with support for some simulators VUnit do not support and extra CLI options. May I ask why you've decided to do that over making contribution to VUnit? We already support SV/UVM, addition of new simulator interfaces and custom CLI options. Are there features that you require which doesn't work well with VUnit?

While you're fully allowed to create derivative projects you're not allowed to remove the license/copyright notice from files you've reused or modified. Mozilla Public License 2.0 requires that you make modifications available (which you have) but that is not something required by the Apache license which means that any downstream project to yours would fail to recognize that obligation. That's not the only difference between the licenses but an important one.

zhajio1988, 2019-05-23:
@LarsAsplund  
hi, thanks for your excellent work. i am sorry about license  issue. i don't have many sense about license.  what can i do for you about license issue?

As for  YASA  test framework.  i mainly worked on SystemVerilog/UVM.  The normal work flow is getting a filelist from design team, then in build.cfg,  create a build which corresponding to a design module or a subsystem. In a build section, includes 
compile_option, sim_option, pre_compile_option, post_compile_option, pre_sim_option, post_sim_option. In group.cfg, actually it's a regression testlist.
by this means, we can run a testcase with specified seed and run multiple times with different seed. or run a regression.
when i come across vunit, i found it supplies many usefull python modules. so i use it in my test framework.

LarsAsplund, 2019-05-23:
> As for YASA test framework. i mainly worked on SystemVerilog/UVM. The normal work flow is getting a filelist from design team, then in build.cfg, create a build which corresponding to a design [...]

It looks like this would be possible with VUnit today. I recommend that you have a look at this [discussion](https://github.com/VUnit/vunit/issues/328). It would save you tons of work and maintenance to go with a VUnit-based solution and if you do VUnit would have improved simulator support (seems like you're working with Incisive and VCS). I've tried to get better support for Incisive/Xcelium for a while (see #501 and #325). Any contributions in that area would be appreciated. Yo can also support https://support.cadence.com/apex/COSCaseDetailPage?CaseID=5000V00001J93GT. 

> hi, thanks for your excellent work. i am sorry about license issue. i don't have many sense about license. what can i do for you about license issue?

If you still decide to proceed with your own project you need to keep the original license and copyright note in the files you're reusing or modifying. New files can have any license compatible with MPL. Apache is such a license.

The normal disclaimer: I'm not a lawyer and cannot give legal advice.

zhajio1988, 2019-05-24:
@LarsAsplund thanks for your suggestion. i will  keep the original license and copyright note in the files. 
i have a question about vuint?
it looks like vunit is a unit test framework. if i implements the features (i mentioned last reply) by Vunit, i need write a lot of pythons scripts. in fact,  some dv engineers prefer to writing init file format。so， could you show me an demo about vunit used in UVM tb regression test?

LarsAsplund, 2019-05-24:
@zhajio1988 Can you provide me with a minimal example of an ini file containing the features you want to support? It would make it easier for me to see how well that approach matches the VUnit scripting features. Once that relation has been establish one can use [configparser](https://docs.python.org/3/library/configparser.html) for the translation

kraigher, 2019-05-26:
VUnit is just a Python library so there is nothing stopping you from adding abstraction layers on top if it. For example if you have an existing .cfg file format you can create a generic run.py file that will parse the .cfg file and run the tests.

zhajio1988, 2019-05-27:
@kraigher could you give an API user guide Vunit supply? so I can try it?

LarsAsplund, 2019-05-31:
First create [a custom command line option] to pass your configuration to the `run.py` script. Then parse it using [configparser](https://docs.python.org/3/library/configparser.html). Finally map the parsed configuration to the corresponding VUnit functionality using the existing APIs. To give you any more help on what that functionality might be we would need an example of such a configuration file.