VUnit/vunit #501: Vunit with Xcelium or Vivado?

petandpal, 2019-05-22:
Hello, I am trying to run Vunit with Xcelium (using the incisive simulator option) but I am seeing some strange issues. First it cannot find the installation root and is looking for Virtuoso. It then goes forward and gets stuck with vunit_lib not found even though it is specified in the cds.lib file. Any help with this?

setenv VUNIT_SIMULATOR incisive
My path points to <installation dir>/XCELIUM1809/tools/bin

[image]

Also are there any plans to have vunit work with Vivado simulator on Linux? Thanks.

1138-4EB, 2019-05-22:
Ref #325.

LarsAsplund, 2019-05-22:
I've spoken to several users of Cadence simulators interested in VUnit but we cannot provide any good support without simulator licenses and that's why we're asking for help in #325. An alternative is to have Cadence to support the project. I've tried to reach out but no success so far but I know that they've used it internally with Xcelium. I encourage you and any other Cadence user to contact them about the need for official VUnit support. I know of companies considering moving to Cadence but the lack of VUnit support is a concern so Cadence is actually losing business because of this.

LarsAsplund, 2019-05-22:
See #209 for Vivado

LarsAsplund, 2019-05-22:
@petandpal Would you be willing to create a VUnit support request on https://community.cadence.com? Then the rest of us can go there and 👍

petandpal, 2019-05-22:
Sure.

LarsAsplund, 2019-05-22:
@petandpal Please post the link when done

petandpal, 2019-05-22:
https://support.cadence.com/apex/COSCaseDetailPage?CaseID=5000V00001J93GT

LarsAsplund, 2019-05-23:
I can't get access to Cadence support unless I have a license, which I don't. The [user community](https://community.cadence.com) is open though.

1138-4EB, 2019-09-18:
Closing this as a duplicate of #325 and #209. The support ticket provided above has been copied to #325.

closed 2019-09-18 by 1138-4EB