VUnit/vunit #504: Incisive errors when compiling Vunit VHDL sources

petandpal, 2019-05-29:
I am trying to run Incisive 15.20 with Vunit but I get syntax errors when compiling the vunit_lib library as shown below:
[image]
[image]

I get a lot more errors if I include the verification components or the OSVVM packages. Is there any easy way to get past these errors (with options) or am I doing something wrong?

Thanks.

LarsAsplund, 2019-05-30:
Initially when the Incisive support was added I remember that there were many issues related to the VHDL standard so it was not possible to to get anything beyond VHDL-93 to work. OSVVM and some of our own packages are VHDL-2008 so the first step would be to set the `VUNIT_VHDL_STANDARD` environment variable to 93. That will exclude such packages from the compilation. However, the offending package is always included and I can't see anything wrong with that line. Can you? I recommend that you report this to Cadance. Maybe you can reduce the problem to a much smaller example that's easier for them to debug but make sure to reference VUnit as I know they've done some testing internally.

petandpal, 2019-06-03:
Lars, 

I tried to get a simple example to work with a Cadence engineer.  They are able to compile it on their side. But it does not work on my system. So perhaps there is an issue with my installation. 

Because I do not have root access to my Python installation directory, I use the third method of installation of vunit  - essentially downloading vunit_hdl-4.0.8.tar.gz from PyPI and untarring it. This does produce all the vunit/vunit_hdl.egg-info directories and files but may be some files are missing or unextracted? Could that explain this?

When I run run setup.py install, it fails because I don't have root access. I tried to use the --install-dir and --prefix options but they don't work. Even putting them inside the setup.cfg file makes no difference. Is there a way to run install into my local directory at all?

1138-4EB, 2019-06-03:
@petandpal, you can try defining envvar `PYTHONPATH=/path/to/your/vunit/repo`.

petandpal, 2019-06-03:
That's what I do currently.

1138-4EB, 2019-06-03:
When you do that, you do not need to install VUnit by running `setup.py`. Just go to your design and execute the corresponding `run.py` directly. Is this what you are doing?

petandpal, 2019-06-03:
Yep.

petandpal, 2019-06-03:
And its not working this way. So I was thinking may be I must be missing something by not installing it the recommended way. So you are confirming that this should not be the source of my issue with compiling to Cadence?

1138-4EB, 2019-06-03:
With either GHDL or ModelSim/QuestaSim, the following procedure works:

``` bash
mkdir /tmp/vunit
#download VUnit to /tmp/vunit with any of git, curl, wget...
export PYTHONPATH=/tmp/vunit
pip3 install colorama
cd /tmp/vunit/examples/vhdl/array
python3 run.py -v
```

Is colorama available on your system? Can you please post the exact error messages that you are getting (in case they are different from the screenshots)?

petandpal, 2019-06-03:
I re-downloaded Vunit into a different directory and set PYTHONPATH to that directory/vunit_hdl-4.0.8.

Now when I run the array example, I see two different errors now when I run Xcelium/Incisive.The errors are the same with Python 2 or 3.  I do have colorama installed. When I run Questasim in the same directory, it works fine. **_Not sure what to do next!!!_**

Here is the Xcelium error snapshot:
[image]

Here is the Incisive error snapshot:
[image]

LarsAsplund, 2019-06-04:
Did you set `VUNIT_VHDL_STANDARD` to `93`? If you do you can't have the context statement in your testbench but rather replace that with the use statements within.

movadaml, 2020-01-20:
I hit this exact issue recently while evaluating vunit.
The following failed with the same error:
- ./run.py
Note that irun_compile_vhdl_file_vunit_lib.args includes "-v200x"
- VUNIT_VHDL_STANDARD=93 ./run.py
Note that irun_compile_vhdl_file_vunit_lib.args includes "-v93"

I was able to work around this by commenting out the prototype for "procedure decode".  A permanent workaround might be to define a wrapper function with a different name for the alias.

After this, I got many compile errors in ansi_pkg.vhd - seems to have an issue with character literals:
`=== Command output: ===
  constant colors_enabled : string_ptr_t := new_string_ptr("0");
                                                            |
ncvhdl_p: *E,ILSTCH (xxx/lib64/python3.6/site-packages/vunit/vhdl/logging/src/ansi_pkg.vhd,76|60): illegal character in string literal - '0' [7.3.1].
    return get(colors_enabled, 1) = '1';
                                      |
ncvhdl_p: *E,IDENRD (xxx/lib64/python3.6/site-packages/vunit/vhdl/logging/src/ansi_pkg.vhd,80|38): identifier ('1') is not visible, as it is directly visible via use clauses from more than one design unit. This results in conflict. This object is defined at:
STD.STANDARD
STD.STANDARD
VUNIT_LIB.STRING_PTR_PKG
.
        while idx <= msg'high and msg(idx) /= 'm' loop
                                                |
ncvhdl_p: *E,IDENRD (xxx/lib64/python3.6/site-packages/vunit/vhdl/logging/src/ansi_pkg.vhd,126|48): identifier ('m') is not visible, as it is directly visible via use clauses from more than one design unit. This results in conflict. This object is defined at:
STD.STANDARD
VUNIT_LIB.STRING_PTR_PKG
.
      if msg(i) = 'm' then
                    |
ncvhdl_p: *E,IDENRD (xxx/lib64/python3.6/site-packages/vunit/vhdl/logging/src/ansi_pkg.vhd,143|20): identifier ('m') is not visible, as it is directly visible via use clauses from more than one design unit. This results in conflict. This object is defined at:
STD.STANDARD
VUNIT_LIB.STRING_PTR_PKG
.
    assert false report "incomplete color escape did not end with 'm'";
                         |
ncvhdl_p: *E,ILSTCH (xxx/lib64/python3.6/site-packages/vunit/vhdl/logging/src/ansi_pkg.vhd,148|25): illegal character in string literal - 'i' [7.3.1].
    assert false report "incomplete color escape did not end with 'm'";
`
- irun(64): 15.20-s020: (c) Copyright 1995-2017 Cadence Design Systems, Inc.
- pip show vunit-hdl
Name: vunit-hdl
Version: 4.3.0

c-thaler, 2022-08-16:
> illegal character in string literal
> identifier ('1') is not visible, as it is directly visible via use clauses from more than one design unit. This results in conflict. This object is defined at: STD.STANDARD STD.STANDARD VUNIT_LIB.STRING_PTR_PKG .

These errors are somehow caused by the following lines in "string_ptr_pkg.vhd":
```
alias  val_t  is character;
alias  vec_t  is string;
```
I commented them out and replaced their occurences in "string_ptr_pkg.vhd" with "character", resp. "string". That fixed  "ansi_pkg.vhd" for me.

LarsAsplund, 2022-08-31:
@c-thaler What version of Incisive/Xcelium are you using?

c-thaler, 2022-09-01:
@LarsAsplund I'm using Incisive 9.2. Which is from around 2011.

I did some more investigations on this. And it was actually only the alias of the character that caused the error. I replaced it with `subtype val_t  is character;`.

To make VUnit work with Incisive 9.2, I had to fix some more things. You can find them in the following two commits on my fork:
https://github.com/tes-gfx/vunit/commit/9fbcc6b0ea9e5af0fe825887ff2edb15032350d9
https://github.com/tes-gfx/vunit/commit/9ee8bec3ab3b0d0056494dec3ce5092de8f32aaa