VUnit/vunit #506: merge_coverage fails with too long argument (on Windows).

pStefanNilsson, 2019-06-03:
The merge_coverage method (in modelsim_interface.py) will crash when ran with too many test cases to merge.

```
Traceback (most recent call last):
  File "C:\Python\Python35-32\lib\site-packages\vunit\ui.py", line 794, in main
    all_ok = self._main(post_run)
  File "C:\Python\Python35-32\lib\site-packages\vunit\ui.py", line 838, in _main
    all_ok = self._main_run(post_run)
  File "C:\Python\Python35-32\lib\site-packages\vunit\ui.py", line 883, in _main_run
    post_run(results=Results(simulator_if))
  File "./run.py", line 54, in post_run
    results.merge_coverage(file_name="merged.ucdb") # Fails if too many files to merge...
  File "C:\Python\Python35-32\lib\site-packages\vunit\ui.py", line 1904, in merge_coverage
    self._simulator_if.merge_coverage(file_name=file_name, args=args)
  File "C:\Python\Python35-32\lib\site-packages\vunit\modelsim_interface.py", line 351, in merge_coverage
    vcover_merge_process = Process(vcover_cmd, env=self.get_env())
  File "C:\Python\Python35-32\lib\site-packages\vunit\ostools.py", line 119, in __init__
    creationflags=subprocess.CREATE_NEW_PROCESS_GROUP)
  File "C:\Python\Python35-32\lib\subprocess.py", line 950, in __init__
    restore_signals, start_new_session)
  File "C:\Python\Python35-32\lib\subprocess.py", line 1220, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 206] The filename or extension is too long

Exception ignored in: <bound method Process.__del__ of <vunit.ostools.Process object at 0x03BE50B0>>
Traceback (most recent call last):
  File "C:\Python\Python35-32\lib\site-packages\vunit\ostools.py", line 241, in __del__
    self.terminate()
  File "C:\Python\Python35-32\lib\site-packages\vunit\ostools.py", line 216, in terminate
    if self._process.poll() is None:
AttributeError: 'Process' object has no attribute '_process'

```

It looks like that when the vcover_cmd string exceeds 32768 characters the underlying subprocess call will crash.

Depending on test names, path, etc we run into this problems at approx 200 test cases.

One possible workaround could be to use the vcover merge -inputs flag and put the path to all the ucdb-files to be merged in a text file instead of appending them directly to the vcover_cmd string.

LarsAsplund, 2019-06-03:
Have you tried your workaround to see if it works? The error message

> FileNotFoundError: [WinError 206] The filename or extension is too long

Suggests something else but maybe it's misleading.

kraigher, 2019-06-03:
We had this problem with RivieraPRO a while back, the fix was to write the files in a TCL script and run the script instead of passing them on command line. If you are proficient enough in Python you can attempt to make a PR of this yourself. 

The relevant code is here:
https://github.com/VUnit/vunit/blob/master/vunit/modelsim_interface.py#L331

Similar solution for RivieraPRO:
https://github.com/VUnit/vunit/blob/master/vunit/rivierapro_interface.py#L348

pStefanNilsson, 2019-06-04:
@LarsAsplund 
Yes, I belive the error message is somewhat missleading. Some [searching](https://stackoverflow.com/questions/2381241/what-is-the-subprocess-popen-max-length-of-the-args-parameter) the error message  indicated that it is an limitation on 32768 characters in the[ Win32 API](https://docs.microsoft.com/en-us/windows/desktop/api/processthreadsapi/nf-processthreadsapi-createprocessa). My own experiments confirmed this.

@kraigher 
I think I can handle the python if you give me a quick intro on the recomended workflow for a PR.

pStefanNilsson, 2019-06-04:
Sorry, newbie :)

LarsAsplund, 2019-06-04:
@pStefanNilsson Basically you start by [forking](https://help.github.com/en/articles/fork-a-repo) our repository to create a copy on which you can make the fix. Once that is done you create a [pull request](https://help.github.com/en/articles/creating-a-pull-request-from-a-fork) from that fork which will allow us to review your work. When your updates have been reviewed and accepted we will merge the changes into VUnit's main repository.

kraigher, 2019-06-12:
You fixed this with the PR

closed 2019-06-12 by kraigher