VUnit/vunit #517: Mocking: check_only_log with any message

ttobsen, 2019-06-13:
In some cases, I just want to check if a log message occures with a given `log_level_t`, but I don't want to verify the content of the message itself. For exammple I just want to check that a error has raised, but I don't care what the content of the error message was.

I'm not sure if this is currently possible, because the `check_only_log` requires the `msg` parameter. If not, maybe it makes sense to `overload check_log` and `check_only_log` with dropping the `msg` parameter?

The main use case for me is, that I want to check that an error has raised, which has a pretty complex message. I'm not interested at all, if the single chars are as expected, I just want to ensure that the error raised.

kraigher, 2019-06-15:
It is not possible at the moment but it would be a good feature to have. Maybe by adding a default msg to the check procedures with the semantic that the message is not checked. This is the way it works with the log time field.

ttobsen, 2019-06-15:
Alright, but there is ine thing why I would suggest to overload the function:

The `msg` parameter is not at the end of the list where the default parameters start. The `msg` and the `log_level` would have to be twisted, which would break compatibility for existing test environments. If not twisting, the function call would look like 
`check_log(my_logger, msg => open, error)`

If you don't mind, I would open a PR and maybe you can help me a little bit?

kraigher, 2019-06-15:
ok go ahead

LarsAsplund, 2019-06-15:
A common mistake (at least for me) when verifying a verification component is that I write a test case which I think will trigger a specific error check but instead it fails too early on another error check. Without checking the message there is a risk that you would fail to discover that mistake. Providing a procedure allowing you to ignore the message may cause people to take a bad decision an go for the "lazy" solution. Another approach that would mitigate that risk is to supply a function allowing you to get the log object and then you can do whatever you like with its contents. For example you can assure that your complex error message at least contains a known static substring using `find` from the `string_ops` package. Makes your test a bit more solid. It's not a one-liner but it's a solution that provides some "protection" for the most common use case.

ttobsen, 2019-06-15:
I understand your doubts. The problem is, that VHDL doesn't provide complex string operations. The best and cleanest option would be, to have something like regular expressions (or at least wildcard support for * and ?). But this is a lot of work to implement.

But having a "lazy" option at least with support of checking a substring using the `find` function, could be a good compromise as a first iteration.

LarsAsplund, 2019-06-16:
> But having a "lazy" option at least with support of checking a substring using the find function, could be a good compromise as a first iteration.

Yes, I think so. We can always add more string operations in the future. `find` is better than nothing and you still have the option to completely ignore the message if you want to.