VUnit/vunit #532: VHDL circular dependency when an architecture maps to another architecture using the same entity

rugebiker, 2019-08-29:
Hi,
I have an error of circular dependency when two architectures use the same entity, and one of them maps to the other one. Attached is a small example testbench where I get this error.

**Description of files:**
**run.py** : Simple, just adds files and launches the test
**tb_circular_dependency** : Empty test, just for example. Maps to entity(architecture1).
**entity_architecture1.vhd** : This file contains an entity and an architecture of that entity.
**architecture2.vhd** : This file contains a second architecture of the same entity provided in the previous mentioned file.
architecture1 (the one on the first file) contains a direct declaration of a component which is entity(architecture2).

The error I get is:
```
  ERROR - Found circular dependency:
~/Projects/circular_dependency/entity_architecture1.vhd ->
~/Projects/circular_dependency/architecture2.vhd ->
~/Projects/circular_dependency/entity_architecture1.vhd
```

Expected behaviour: There is no real VHDL circular dependency here. Both ModelSim (standalone simulation) and Vivado (Simulation and Synthesis) work correctly.

Example project:
[circular_dependency.tar.gz](https://github.com/VUnit/vunit/files/3554490/circular_dependency.tar.gz)
just run:
`python run.py`

1138-4EB, 2019-08-29:
I think there is a circular dependecy between files, indeed. File `entity_architecture1.vhd` requires `architecture2.vhd` to be analyzed first, and viceversa. You might argue that VUnit could identify the units in the files, and try to break the circle by considering `entity_architecture1.vhd` to be two separate units/files. But VUnit does not split/merge/reorganize the VHDL code.

The expected structure is:

- Split `entity_architecture1.vhd` to `entity.vhd` and `architecture1.vhd`.
- Or, interchange architectures, so you get `entity_architecture2.vhd` and `architecture1.vhd`.

Any of these approaches should avoid the error.

go2sh, 2019-08-30:
That might not true. There is no real circular dependency. The architecture of the instantiation is only evaluated during elaboration, I guess.

1138-4EB, 2019-08-30:
> That might not true. There is no real circular dependency. The architecture of the instantiation is only evaluated during elaboration, I guess.

You are correct, from a VHDL point of view. Precisely, none of the simulators fail. @rugebiker tested it with ModelSim and Vivado, and I tested it with GHDL (although I had to fix `tb_circular_dependecy.vhd`). Thus, my previous comment is about how VUnit handles dependecy, which is "per file".

go2sh, 2019-08-30:
AFAIK vunit uses the dependency scanner to generate the compile order. In the presented case, the compile order can be solved.
 * entity_architecture1.vhd
 * architecture2.vhd
 * tb_circular_dependency.vhd

1138-4EB, 2019-08-30:
See https://github.com/VUnit/vunit/blob/master/vunit/dependency_graph.py#L65-L71. I think that it raises an exception as soon as it visits a node twice, where a node is a file.

go2sh, 2019-08-30:
Yes, the problematic code is in [here](https://github.com/VUnit/vunit/blob/master/vunit/project.py#L205-L227). There has to be a skip, if the entity referenced is in the currently analyzed file.

rugebiker, 2019-09-03:
Indeed. Maybe it is an idea to completely split the entities and architectures into different trees, but I can imagine it would be a lot of redesign.

kraigher, 2019-09-14:
Maybe we can remove this dependency. I just vaguely remember that we added the dependency due to a problem we found. Maybe it was when an architecture is referenced explicitly in a configuration.