VUnit/vunit #545: module '' previously defined

oscargus, 2019-09-19:
There seems to be a problem when including a file that gives the warning:

`module '' previously defined`

For quite some time, it seemed like giving an absolute path to a file giving that warning caused it, but with a slightly different file system/path, that is no longer the significant difference. (University computer environment, so different file systems/paths for students and employees. Seems to be the case for the student one, although the VUnit version etc should be the same. A similar issue sometimes show up on the employee side as well, but I have not been able to really figure out the conditions for a failure.)

Also, it is not really obvious where this empty module is defined in the source. The source file in question is `altera_primitives.v`, so the basic Intel/Altera generic primitives (therefore I cannot really provide any more details...), but it does not seem like there are any modules with empty names (and once using a soft link or copying the file to the same directory, the problem disappears, if it ever showed up, while the warning remains). The test bench in question either instantiates the original design or the synthesized design based on a generic (there are enums in the VHDL interface so two different instantiations) and the consequence of the issue is that the synthesized version ends after the test bench in the compile order list. Sometimes, it works, sometimes the order comes out wrong. The only really clear connection is the warning in that file (there is also another file from the Altera/Intel library included which doesn't seem to cause any errors, and no warnings). The used version of `altera_primitives.v` is Quartus Prime 17.1.0 Build 590 10/25/2017 and although not the same version there is a more or less similar version (in terms of how modules are written etc) in https://github.com/asicguy/gplgpu/blob/master/hdl/sim_lib/altera_primitives.v which gives the same type of warning.

I realize that this is a more than vague issue, but for what it is worth. Maybe someone with a better knowledge of the code base can directly realize the potential issue. A guess would be that the empty module name detection may be a cause somehow. I cannot really figure out where it may come from though (which leads to a request to possibly provide a bit more details, like line number, for the parsing warnings/errors).

kraigher, 2019-09-19:
It seems like you have two problems, One being the warning that the module '' is previously defined. The other that the dependency between DUT and testbench in the compile order is missing. Are you able to create a [minimal working example](https://en.wikipedia.org/wiki/Minimal_working_example) for the second problem?

Also as a work around VUnit supports adding manual dependencies between files using the Python API.

oscargus, 2019-09-19:
I've only encountered the second problem while simultaneously having the first. Or rather, that seems to be the only (sometimes) crucial difference I can identify. But then, we've only recently started to use VUnit.

I'll try to see if I can trigger it more consistently and get back with a minimal working example (I'm aware it is missing, but just though I'd write down what I've come up with so far in the hope that there may be some "obvious" insights). While consistently reproducible once it shows up, I've not been able to fully figure out what is required for it to show up...

kraigher, 2019-09-19:
I can take a look on the previously defined module problem and see if the fix could be related to dependency scanning. Dependency scanning for Verilog is a bit more complex than VHDL since there is a preprocessor. VUnit must preprocess the code before it can extract dependencies which mean you need to set the correct include dirs and `defines when adding the source files using the Python API. Also encrypted code cannot be scanned for dependencies obviously.

kraigher, 2019-09-19:
The problem occurs because `global` and `soft` are keywords in the latest SystemVerilog standard. The VUnit tokenizer stores Tokens as kind, value pairs. For identifiers the value is the name of the identifier. For keywords the value is empty to save space since the each keyword has a unique kind. Thus VUnit will scan a module with the an empty name twice in the altera_primitives.v for `global` and `soft`. I will push a fix for it soon.

kraigher, 2019-09-19:
I pushed a fix to master now for the duplicate '' module.

oscargus, 2019-09-23:
Thanks! Will try it out!

oscargus, 2019-09-24:
I can confirm that the warning is removed. Will see if I manage to get the error. Thanks again!